project-chip/connectedhomeip #29082: Question about CHIPTool libraries on Android

OlivierGre, 2023-09-06:
I'm using a Google Pixel 6 phone running Android 13 and I'm looking at Matter commissioning.

Android native code has some code to detect the BLE advertizing of Matter devices and to perform the Matter Commissioning.
On my smartphone, I have installed Google Home Sample App which also uses CHIPTool libraries (libCHIPController.so). 

I'm wondering how it is working because we basically have twice the same code. For example, I can see in Android Logcat that both (the native code and google homesampleapp) are sending a PBKDFParamRequest:

Here is the native trace:
`2023-09-05 16:54:00.709 15137-15137 native                  pid-15137                            I  I0000 00:00:1693925640.708980   15137 chip_logging.cc:17] CHIP: EM: <<< [E:13931i S:0 M:193057879] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)`

And here is the trace from "libCHIPController.so" embedded in homesampleapp:
`2023-09-05 16:54:26.377 21276-21276 EM                      com.google.homesampleapp.default     D  <<< [E:26031i S:0 M:6547496] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest) --- Len 52`

The same thing would happen if I was using Matter CHIPTool app.

Could you please explain me how those "duplicate" codes are living together and how you avoid interferences?

Thank you!

OlivierGre, 2023-09-06:
I can see in the logcat that, when my Matter device is switched on, the BLE advertizing is detected by android native code and Matter commissioning is done. 

Then I can see that "onCommissioningRequested" of google.homesampleapp is called and another commissioning is done for the sample app. A comment in the source code says: 
` // Perform commissioning on custom fabric for the sample app.`

OlivierGre, 2023-09-06:
If my understanding is correct, the first commissioning (the native one) is done by Google Play service and the second commissioning is done by Google Home Same App. 

The first commissioning brings the network connectivity to the Matter device (it can communicates over Thread). 
The second commissioning allows Google Home Same App to communicate with the device. This second commissioning relies on the existing network (Wifi + Thread) to communicate with the device.

yunhanw-google, 2025-04-23:
Yes, libCHIPController.so has been put into Google play service, you can use google home API to commission your matter device.

You can also compile android chip-tool where  libCHIPController.so has been put inside this app, you can also use this tool to commission the matter device which bypass google play service.

I am closing this one now, feel free to reopen if there is any further questions.