project-chip/connectedhomeip #29086: [YAML]  EqualityCommands cluster command UnsignedNumberEquals fails for NULL value comparison.

manjunath-grl, 2023-09-06:
### Reproduction steps

Summary:
According to the test plan, step 2 of the TCCM 3.2 test checks whether the `StartUpMode` attribute has the value integer or null, and if startup_mode_dut is null, proceeds to step 3. When executed in YAML got the value of StartUpMode as `None` using the EqualityCommands cluster command UnsignedNumberEquals, we tried to compare the value as `null` instead of `true` got `false` as the response.

Reproduction steps:
Terminal 1:
Launch `./chip-all-clusters-app`

Terminal 2:
Step 1: Pair the DUT
`./out/chip-tool/chip-tool pairing code 0x12344321 MT:-24J0AFN00KA0648G00`
Step 2: Run test TCCM_3_2
`./scripts/tests/yaml/chiptool.py tests Test_TC_TCCM_3_2`

Effected tests:
TC-TCCM-3.2
TC-TCCM-3.3
TC-RVCCLEANM-3.3
TC-RVCRUNM-3.3

Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:UnsignedNumberEquals_Fails_for_NULL_Value_Comparsion

### Bug prevalence

Everytime when executed

### GitHub hash of the SDK that was being used

6b708ab29de982d9fa0363d6ce8a2b9a9840b31a

### Platform

other

### Anything else?

Testplan screenshot:
[image]

YAML screenshot:
[image]

Execution logs & screenshot:
[image]
[image]
[TCCM32_Execution_logs.zip](https://github.com/project-chip/connectedhomeip/files/12536421/TCCM32_Execution_logs.zip)

bzbarsky-apple, 2023-09-06:
The schema for UnsignedNumberEquals is:
```
    <command source="client" code="2" name="UnsignedNumberEquals" response="EqualityResponse">
      <arg name="Value1" type="int64u"/>
      <arg name="Value2" type="int64u"/>
    </command>
```
so `null` would not be a valid value for its arguments.  I am pretty surprised that this did not just throw an error; feels like it should have....

manjunath-grl, 2023-09-06:
> The schema for UnsignedNumberEquals is:
>
> ```
>     <command source="client" code="2" name="UnsignedNumberEquals" response="EqualityResponse">
>       <arg name="Value1" type="int64u"/>
>       <arg [...]

In YAML I am not expected any response value that's y its not failing. If I expect true it fails.
[image]

vivien-apple, 2023-09-07:
I am not sure of why we do check for `None` here. I can see that this is me that have added the change but I don't remember why on top of my head... I made https://github.com/project-chip/connectedhomeip/pull/29107 to revert this particular change.

PR #29107 (2023-09-12): [MatterYamlTests] null values that are saved are not substituted when…
… the key used to saved them is used

fix #29086

It looks like a regression that I have introduced while working on the pics checker. I have added a test in `TestSaveAs.yaml`
closed 2023-09-12 by mergify[bot]