project-chip/connectedhomeip #29087: [1.0] Error retrieving data from clusterId - No memory

mustafayuce33, 2023-09-06:
### Reproduction steps

I started testing my matter bridge for 2 days. Today, devices say "No responding" on Apple Home. And when I take a look at the logs I see a lot of messages in the log even If I control a single device and in one of the message it says `Error retrieving data from clusterId: 0x0000_0102, err = src/system/TLVPacketBufferBackingStore.cpp:88: CHIP Error 0x0000000B: No memory`. After I restart my service everything started working again. Here is the log file attached when I turn off (or) a bulb, which means it is just a single command. 

I think there is something wrong with having such a huge log records for just a single command.  

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/tree/1084dd53d50b54e52cc0d91a0f39f84dc68b7039

### Platform

raspi

### Type

Core SDK Crash

mustafayuce33, 2023-09-06:
Can it be this issue related with this https://github.com/project-chip/connectedhomeip/issues/28434?

mustafayuce33, 2023-09-11:
@bzbarsky-apple Any idea?

bzbarsky-apple, 2023-09-11:
The "No memory" errors are expected given that this is doing a chunked read; that just indicates current chunk is full.  See the immediate next line in the log:
```
Sep 06 12:41:20 test-serialnumber test-matter[1486]: CHIP:DMG: <RE:Run> We cannot put more chunks into this report. Enable chunking.
```

Past that, I have no idea why the client would be claiming "No responding".  That's something that would need client-side logs to diagnose.

As for why you are seeing reporting of all sorts of attributes after "just a single command"... that depends on exactly what code is running on the server.    But it generally looks like there is priming going on for a new subscription in parallel with whatever happens due to the command... or something has marked _all_ attributes on the server dirty.

closed 2023-09-18 by mustafayuce33