project-chip/connectedhomeip #29089: [Build] Cannot build py controller on Raspberry Pi 4B running Ubuntu Server 22

rocveralfre, 2023-09-06:
### Build issue(s)

After setting up the pi following the README file in the corresponding folder and after having to install gn and ninja (not written in the README), here I am facing an issue in the build process but I have no clue what that may be..

```shell
veralfre@veralfrepi:~/connectedhomeip$ ./scripts/build_python.sh -m platform -i separate
Input values: chip_detail_logging = false , chip_mdns = "platform", enable_pybindings = false, chip_case_retry_delta="", pregen_dir=""

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (4.3s)
  Setting up Project actions........skipped (0.1s)
  Setting up Python environment.....[|]
ninja: Entering directory `/home/veralfre/connectedhomeip/.environment/gn_out'
[1/545] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//build/toolchain/linux:linux_arm_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --capture-output --module piptools --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/matter_build_venv._compile_requirements_metadata_path_list.txt -- compile --resolver=backtracking --allow-unsafe --output-file python/gen/matter_build_venv/compiled_requirements.txt python/gen/matter_build_venv/generated_requirements.txt ../../scripts/setup/requirements.build.txt
FAILED: python/gen/matter_build_venv/compiled_requirements.txt 
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//build/toolchain/linux:linux_arm_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --capture-output --module piptools --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/matter_build_venv._compile_requirements_metadata_path_list.txt -- compile --resolver=backtracking --allow-unsafe --output-file python/gen/matter_build_venv/compiled_requirements.txt python/gen/matter_build_venv/generated_requirements.txt ../../scripts/setup/requirements.build.txt
    error: subprocess-exited-with-error
    
    × Getting requirements to build wheel did not run successfully.
    │ exit code: 1
    ╰─> [54 lines of output]
        running egg_info
        writing lib/PyYAML.egg-info/PKG-INFO
        writing dependency_links to lib/PyYAML.egg-info/dependency_links.txt
        writing top-level names to lib/PyYAML.egg-info/top_level.txt
        Traceback (most recent call last):
          File "/home/veralfre/connectedhomeip/.environment/gn_out/python-venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
            main()
          File "/home/veralfre/connectedhomeip/.environment/gn_out/python-venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
            json_out['return_val'] = hook(**hook_input['kwargs'])
                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
          File "/home/veralfre/connectedhomeip/.environment/gn_out/python-venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
            return hook(config_settings)
                   ^^^^^^^^^^^^^^^^^^^^^
          File "/tmp/pip-build-env-dx3u84_e/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 355, in get_requires_for_build_wheel
            return self._get_build_requires(config_settings, requirements=['wheel'])
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
          File "/tmp/pip-build-env-dx3u84_e/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 325, in _get_build_requires
            self.run_setup()
          File "/tmp/pip-build-env-dx3u84_e/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 341, in run_setup
            exec(code, locals())
          File "<string>", line 288, in <module>
          File "/tmp/pip-build-env-dx3u84_e/overlay/lib/python3.11/site-packages/setuptools/__init__.py", line 107, in setup
            return distutils.core.setup(**attrs)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
          File "/tmp/pip-build-env-dx3u84_e/overlay/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 185, in setup
            return run_commands(dist)
                   ^^^^^^^^^^^^^^^^^^
          File "/tmp/pip-build-env-dx3u84_e/overlay/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
            dist.run_commands()
          File "/tmp/pip-build-env-dx3u84_e/overlay/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
            self.run_command(cmd)
          File "/tmp/pip-build-env-dx3u84_e/overlay/lib/python3.11/site-packages/setuptools/dist.py", line 1233, in run_command
            super().run_command(command)
          File "/tmp/pip-build-env-dx3u84_e/overlay/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
            cmd_obj.run()
[... 131 lines cut ...]
  File "/home/veralfre/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 959, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/veralfre/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 951, in main
    return EnvSetup(**vars(parse())).setup()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/veralfre/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 558, in setup
    result = step(spin)
             ^^^^^^^^^^
  File "/home/veralfre/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 747, in virtualenv
    if not virtualenv_setup.install(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/veralfre/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 433, in install
    install_packages(gn_target)
  File "/home/veralfre/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 414, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['ninja', '-C', '/home/veralfre/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.
```

I am running the OS on a 128 GB sd card, therefore I do not expect it to be an Out of Space issue, as reported in some other tasks.
I need to build the py controller as It may become part of a bigger solution, where the chip-controller repl is not feasible for the use case..

### Platform

raspi

andy31415, 2024-10-24:
The `AttributeError: cython_sources` leads me to believe that you need some python-dev modules here.

andy31415, 2024-10-24:
@rocveralfre could you provide instructions to reproduce ? Is this a pure Ubuntu 22 on RPi? Could you try out Ubuntu 24 as well since I believe we started using that in latest master builds and tests instead of 22 (even though 22 should also work...)

denravonska, 2024-12-10:
I have the same thing on Ubuntu  24.04 x64. No idea how to reproduce it :(