project-chip/connectedhomeip #29115: [Test_TC_IDM_1_2] Unknown clusters lead to harness failures in matter_chip_tool_adapter

bzbarsky-apple, 2023-09-07:
### Test issue(s)

When trying to run the Test_TC_IDM_1_2 YAML test against all-clusters-app, I get failures like so:
```
2023-09-07 15:13:16.794 ERROR   15:13:16.713 - TEST ERR  :   File "examples/chip-tool/py_matter_chip_tool_adapter/matter_chip_tool_adapter/decoder.py", line 113, in __translate_names
2023-09-07 15:13:16.794 ERROR   15:13:16.713 - TEST ERR  :     raise KeyError(
2023-09-07 15:13:16.794 ERROR   15:13:16.713 - TEST ERR  : KeyError: 'Error: The cluster (0x7) attribute (0xfff9) definition can not be found. Please update the cluster definition.'
2023-09-07 15:13:16.794 ERROR   15:13:16.736 - TEST OUT  :              7. Running Get the list of available cluster commands on the device (all endpoints)
```

It seems like his wildcard read is getting dumped into code that expects to know about all clusters, which we just do not.  This specific cluster is draft, but this would presumably also fail for manufacturer-specific things.

### Platform

core (please add to version below)

cecille, 2023-09-08:
Is this happening on ToT? Why would this not cause CI failures?

cecille, 2023-09-08:
Is this blocking something in SVE right now?

bzbarsky-apple, 2023-09-08:
> Is this happening on ToT?

Yes.

> Why would this not cause CI failures?

Because for several weeks now we have not been running YAML tests in CI, effectively.  I ran into this while trying to fix that in #29096 and I disabled this YAML test as part of that PR, since it fails.

> Is this blocking something in SVE right now?

No.  SVE is using `src/python_testing/TC_IDM_1_2.py` as I understand it.  Raising the question: Why does `src/app/tests/suites/certification/Test_TC_IDM_1_2.yaml` even exist?

cecille, 2023-09-09:
> > Is this happening on ToT?
> 
> Yes.
> 
> > Why would this not cause CI failures?
> 
> Because for several weeks now we have not been running YAML tests in CI, effectively. I ran into this while trying to fix that in #29096 and I disabled this YAML test as part of that PR, since it fails. This is the issue tracking the fact that it's disabled.

Oh, that's special. :( 
> 
> > Is this blocking something in SVE right now?
> 
> No. SVE is using `src/python_testing/TC_IDM_1_2.py` as I understand it. Raising the question: Why does `src/app/tests/suites/certification/Test_TC_IDM_1_2.yaml` even exist?

I'm not sure, but it seems like the easiest thing to do to unblock this is to delete that test, and just open an issue to fix wildcard reads in the yaml runner, if we still care about wildcards in yaml.

bzbarsky-apple, 2023-09-11:
> Oh, that's special. :(

Hey, at least now we have CI that will fail if we we _not_ running YAML tests, I hope... ;)

> the easiest thing to do to unblock cert/fix CI is to delete that test, 

That seems perfectly reasonable to me if we are not using this "certification" test for cert.  @manjunath-grl @raju-apple

PR #29469 (2023-09-28): Remove TC-IDM-1.2 yaml
This test has a functional python implementation that is fully automated. The yaml is no longer required, and doesn't work anyway.

Fixes: #29115
commit d1fa6ee5 (2023-09-28): Remove TC-IDM-1.2 yaml (#29469)

This test has a functional python implementation that is fully
automated. The yaml is no longer required, and doesn't work anyway.

Fixes: #29115
closed 2023-09-28 by mergify[bot]