project-chip/connectedhomeip #29118: third_party github clone very slow

tpitman, 2023-09-08:
### Reproduction steps

I am following this page:

[https://github.com/espressif/connectedhomeip/blob/v1.1-branch/docs/guides/BUILDING.md#prerequisites](https://github.com/espressif/connectedhomeip/blob/v1.1-branch/docs/guides/BUILDING.md#prerequisites)

It is in the middle of cloning the third_party folders.

It is on the mbed-mcu-boot section.

My download speeds are around 20.00 KiB/s, yes K not even M.

I have very fast internet and the main part of the github clone went super fast.

Doesn't all of this stuff come from github? Should't I get the same speed for all of it?

Can anyone tell me why this would go so slow?

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

46070d75e06058a2e6a525d8be71b9b872178514

### Platform

darwin, esp32

bzbarsky-apple, 2023-09-08:
Checking out all the submodules for all possible platform builds takes a while, and involves some quite large downloads, some of which can be slow, as you noticed.  

What you probably want to do is clone without submodules and then run `scripts/checkout_submodules.py` which will check out the ones relevant to your build OS.  You can pass `--platform` arguments to that script to check out submodules relevant to building for a specific platform.

We should probably get our docs updated.

tpitman, 2023-09-08:
Thank you for the reply. Maybe you can guide me what exactly to pass to the checkout_submodules.py script. I want to do matter development for the esp32 platform. If there is documentation no that it would be great.

What I don't understand about the slow download is why they are slow if they are all github repos.

I am talking about repos that say they are only a few megabytes and yet they download at speeds in the Kilobytes and take 10-15 minutes.

Shouldn't all downloads from github go the same speed?

tpitman, 2023-09-08:
I did the script using --platform esp32 and that didn't do anything so maybe all submodules are already downloaded.

So next step is to run bootstrap.sh. I try to run it and it gives and error:

zsh: permission denied

I tried using sudo and that just says it can't find the script even though I give it the path.

Next?

bzbarsky-apple, 2023-09-08:
> Shouldn't all downloads from github go the same speed?

That's a question for github...  They could be throttling things on their end, for example.

> so maybe all submodules are already downloaded.

If you already did a recursive checkout, yes.

>  I try to run it and it gives and error

How are you running it, exactly?  You want `source scripts/bootstrap.sh` or `. scripts/bootstrap.sh`.

tpitman, 2023-09-08:
For the last one I did a dummy. I have done this clone several times and each time used chmod +x on the script before running it. This time I forgot that and also found using source instead of modifying the permissions.

Now I am getting a different error when running the bootstrap.sh and have a different issue for that.

I will close this because it seems github can throttle if they want and I didn't know that.

closed 2023-09-08 by tpitman