project-chip/connectedhomeip #29120: [1.0.0.2] [BUG] Commissioning fails with "AttestationVerification" code 101

vineelasmile, 2023-09-08:
### Reproduction steps

1. We are building our own matter controller on top on v1.0.0.0 
2. It successfully controls ESP32 led light -on/off etc.
3. When trying with a off the shelf product : Orein Matter Light bulb(ordered from amazon), I see attestation failure such as this:
```
09-08 05:09:31.559 10658 10672 D CTL     : Performing next commissioning step 'AttestationVerification'
09-08 05:09:31.559 10658 10672 D CTL     : Verifying attestation
09-08 05:09:31.609 10658 10672 E CTL     : Failed in verifying 'Attestation Information' command received from the device: err 101. Look at AttestationVerificationResult enum to understand the errors
....
09-08 05:09:31.609 10658 10672 D CTL     : Error on commissioning step 'AttestationVerification': '../../src/controller/CHIPDeviceController.cpp:1075: CHIP Error 0x000000AC: Internal error'
09-08 05:09:31.609 10658 10672 E CTL     : Failed to perform commissioning step 8
09-08 05:09:31.609 10658 10672 D CTL     : Going from commissioning step 'AttestationVerification' with lastErr = '../../src/controller/CHIPDeviceController.cpp:1075: CHIP Error 0x000000AC: Internal error' -> 'Cleanup'
09-08 05:09:31.610 10658 10672 D CTL     : Performing next commissioning step 'Cleanup' with completion status = '../../src/controller/CHIPDeviceController.cpp:1075: CHIP Error 0x000000AC: Internal error'
``` 

Based on https://github.com/project-chip/connectedhomeip/issues/26142 suggestions I have passed the PAA certs from both credentials/development/paa-root-certs and credentials/production/paa-root-certs, and it still fails to find the cert. 

How do I workout where to find the root PAA certificate ? 

### Bug prevalence

all the time

### GitHub hash of the SDK that was being used

v1.0.0.2 tip

### Platform

android

### Type

Platform Issue, Core SDK Interopability Issue

### Anything else?

@bzbarsky-apple tagging you here since you did help in the other issue (https://github.com/project-chip/connectedhomeip/issues/26142). Could point me in the right direction ?

bzbarsky-apple, 2023-09-08:
@vineelasmile It's possible that this is a new PAA that has been added to the DCL but not yet synced into the source tree...  @tcarmelveilleux might be able to point you in the right direction there.

That said, it might be possible to figure out what PAA is involved by looking at what PAI cert the device sends back and what it claims to be signed by.

vineelasmile, 2023-09-08:
@bzbarsky-apple can you give me a code pointer from the matter repo on how to grab the PAI cert the device is sending ?

bzbarsky-apple, 2023-09-08:
Add logging of `response.certificate` to `DeviceCommissioner::OnCertificateChainResponse`?  Just make sure to not get truncated output.

ikodev, 2024-12-21:
Same prob from an ubuntu dist (on rasp5) 
Here are my debug log of chip-tool command, can it helps you ?

```
[1734792786.144] [6460:6462] [DMG] Received Command Response Data, Endpoint=0 Cluster=0x0000_003E Command=0x0000_0001
[1734792786.144] [6460:6462] [CTL] Received Attestation Information from the device
[1734792786.144] [6460:6462] [CTL] Successfully finished commissioning step 'SendAttestationRequest'
[1734792786.144] [6460:6462] [CTL] AutoCommissioner setting attestationElements buffer size 283/283
[1734792786.144] [6460:6462] [CTL] Commissioning stage next step: 'SendAttestationRequest' -> 'AttestationVerification'
[1734792786.144] [6460:6462] [CTL] Performing next commissioning step 'AttestationVerification'
[1734792786.144] [6460:6462] [CTL] Verifying attestation
[1734792786.148] [6460:6462] [CTL] Error on commissioning step 'AttestationVerification': 'src/controller/CHIPDeviceController.cpp:1288: CHIP Error 0x000000AC: Internal error'
[1734792786.148] [6460:6462] [CTL] Failed verifying attestation information. Now checking DAC chain revoked status.
[1734792786.148] [6460:6462] [CTL] Commissioning stage next step: 'AttestationVerification' -> 'AttestationRevocationCheck'
[1734792786.148] [6460:6462] [CTL] Performing next commissioning step 'AttestationRevocationCheck' with completion status = 'src/controller/CHIPDeviceController.cpp:1288: CHIP Error 0x000000AC: Internal error'
[1734792786.148] [6460:6462] [CTL] Verifying device's DAC chain revocation status
[1734792786.148] [6460:6462] [CTL] Failed in verifying 'Attestation Information' command received from the device: err 101. Look at AttestationVerificationResult enum to understand the errors
[1734792786.148] [6460:6462] [CTL] Error on commissioning step 'AttestationRevocationCheck': 'src/controller/CHIPDeviceController.cpp:1337: CHIP Error 0x000000AC: Internal error'
[1734792786.148] [6460:6462] [CTL] Going from commissioning step 'AttestationRevocationCheck' with lastErr = 'src/controller/CHIPDeviceController.cpp:1337: CHIP Error 0x000000AC: Internal error' -> 'Cleanup'
[1734792786.148] [6460:6462] [CTL] Performing next commissioning step 'Cleanup' with completion status = 'src/controller/CHIPDeviceController.cpp:1337: CHIP Error 0x000000AC: Internal error'
[1734792786.148] [6460:6462] [CTL] Disarming failsafe on device 0xffffa400b170
```

Thanks a log

bzbarsky-apple, 2025-01-03:
@ikodev You need to make sure you tell chip-tool to use the relevant set of PAAs.