project-chip/connectedhomeip #29124: Inconsistency between spec and SDK for Rotating Device Identifier Algorithm

fmatignon-netatmo, 2023-09-08:
We have noticed an inconsistency between the spec and the implementation regarding the Rotating Device Identifier Algorithm.

In section "5.4.2.4.5. Rotating Device Identifier", the spec states

> The Rotating Device Identifier Algorithm is defined as follows:
> ```
> Rotating Device ID = Rotation Counter || Crypto_KDF(
>                                              inputKey := Unique ID,
>                                              salt:= Rotation Counter,
>                                              info := "RotatingDeviceID",
>                                              len := 128)
> ```
> (where || is the concatenation operation)

This is from Matter 1.1 spec, but is still valid for 1.0 and the main branch of the spec GitHub repository.

The implementation computes a SHA-256 of the unique ID concatenated with the counter as it can be seen in:

https://github.com/project-chip/connectedhomeip/blob/ce66c4da38ac20070c0ab0e8701a3621d79fa831/src/setup_payload/AdditionalDataPayloadGenerator.cpp#L95-L107

The implementation seems to match with a previous version of the spec text, before 1.0 release.

bzbarsky-apple, 2023-09-08:
Looks like this was never updated to https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/1870?  @decenzo @hnnajh

chrisdecenzo, 2023-09-08:
Given that the SDK has shipped and this functionality is in use, we need to be a bit careful here about changing the SDK to ensure product makers who are using it don't have their functionality break. We probably need a compile-time flag to indicate the old versus new algorithm.

Before doing that, can someone comment on whether the spec change on this is materially important to preserve? Eg. is it more secure?

tcarmelveilleux, 2023-09-08:
There is NO standard algorithm for the rotating identifier. The scheme in the spec is merely an example. Different schemes that respect the spec are usable.