project-chip/connectedhomeip #29125: [ESP32] IDF component manager pulls in cbor component for idfv4.4

PSONALl, 2023-09-08:
### Reproduction steps

Cbor is supported in esp-idf v5.0 and above but gets pulled in for IDF v4.4

When I include the esp-insights component to my project i.e, esp-matter

Insights has dependency on cbor component

case 1. On Esp-IDF v4.4- I face duplicate component issue (as there is an existing cbor component in IDF and it again gets pulled in)

case 2. Workaround: Tried i. temperarily removing the component from idf, also tried ii. editing dependencies.lock  but it gives another failure which indicates that the component is missing

Attaching logs for these two test cases below:
My idf_component.yml is here: https://github.com/project-chip/connectedhomeip/blob/master/config/esp32/components/chip/idf_component.yml

[case1.txt](https://github.com/project-chip/connectedhomeip/files/12558384/case1.txt)
[case2.txt](https://github.com/project-chip/connectedhomeip/files/12558385/case2.txt)

### Bug prevalence

EveryTime

### GitHub hash of the SDK that was being used

195ededb0478c545088e714828486f84ea10ec65

### Platform

esp32

bzbarsky-apple, 2023-09-08:
So does https://github.com/project-chip/connectedhomeip/pull/29123 fix this @PSONALl ?

PSONALl, 2023-09-15:
> So does #29123 fix this @PSONALl ?

Yes, It does but it is a workarround