project-chip/connectedhomeip #29129: MacOS bootstrap.sh gives security: SecKeychainSearchCopyNext: The specified item could not be found in the keychain.

tpitman, 2023-09-08:
### Reproduction steps

Follow these instructions: 

[https://github.com/espressif/connectedhomeip/blob/v1.1-branch/docs/guides/BUILDING.md#prerequisites](https://github.com/espressif/connectedhomeip/blob/v1.1-branch/docs/guides/BUILDING.md#prerequisites)

When I run the scripts/bootstrap.sh script I get the following output:

`Downloading and installing packages into local source directory:`
``
`  Setting up CIPD package manager...done (2.1s)`
`  Setting up Project actions........skipped (0.1s)`
`  Setting up Python environment.....done (3.4s)`
`  Setting up pw packages............skipped (0.1s)`
`  Setting up Host tools.............done (0.1s)`
``
`Activating environment (setting environment variables):`
``
`  Setting environment variables for CIPD package manager...done`
`  Setting environment variables for Project actions........skipped`
`  Setting environment variables for Python environment.....done`
`  Setting environment variables for pw packages............skipped`
`  Setting environment variables for Host tools.............done`
``
`Checking the environment:`
``
`security: SecKeychainSearchCopyNext: The specified item could not be found in the keychain.`
`Error during bootstrap--see messages above.`
`Installing pip requirements for all...`

### Bug prevalence

always

### GitHub hash of the SDK that was being used

latest from that link

### Platform

esp32

### Platform Version(s)

latest

bzbarsky-apple, 2023-09-08:
I just tried doing this:
```
git clone git@github.com:espressif/connectedhomeip.git

cd connectedhomeip

git checkout v1.1-branch

. ./scripts/bootstrap.sh
```

That gives me SHA `06a421e706f38c7de7ea6b627c01269d4273785f` and bootstrap fails with a different error from the one you have, because of the ZAP update that happened on September 5 and I am on an ARM mac.  I assume you are on an Intel mac or using a different SHA (which one, then?).  

Anyway, if I fix that problem as in https://github.com/project-chip/connectedhomeip/pull/29132, then bootstrap succeeds.

If it's failing for you @tpitman :

1. What MacOS version is this?
2. On what platform?
3. Does bootstrap on tip of master (e.g. SHA ce66c4da38ac20070c0ab0e8701a3621d79fa831) work for you?  That is, is this a v1.1-only issue?

tpitman, 2023-09-08:
I am on Apple Silicon mac.

I can't remember the initial git clone command I used. It isn't in my history anymore, which is weird.

The SHA is: ce66c4da38ac20070c0ab0e8701a3621d79fa831

I will have to try a new checkout and let you know.

tpitman, 2023-09-08:
I just tried your clone sequence above and got this error while installing packages:

`[P13772 10:34:29.777 resolver.go:180 D] Failed to resolve package fuchsia/third_party/zap/mac-arm64@version:2@v2023.09.01-nightly.1: no such tag`
`Errors:
  failed to resolve fuchsia/third_party/zap/mac-arm64@version:2@v2023.09.01-nightly.1 (line 13): no such tag
/Users/tpitman/connectedhomeip/.environment/cipd/cipd ensure -ensure-file /Users/tpitman/connectedhomeip/.environment/cipd/packages.ensure -root /Users/tpitman/connectedhomeip/.environment/cipd/packages -log-level debug -json-output /Users/tpitman/connectedhomeip/.environment/cipd/packages.json -cache-dir /Users/tpitman/.cipd-cache-dir -max-threads 0`
` `
`None`
`Traceback (most recent call last):`
`  File "/Users/tpitman/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 877, in <module>
    sys.exit(main())
             ^^^^^^`
`  File "/Users/tpitman/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 869, in main
    return EnvSetup(**vars(parse())).setup()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^`
`  File "/Users/tpitman/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 522, in setup
    result = step(spin)
             ^^^^^^^^^^`
`  File "/Users/tpitman/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 635, in cipd
    if not cipd_update.update(
           ^^^^^^^^^^^^^^^^^^^`
`  File "/Users/tpitman/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/update.py", line 361, in update
    subprocess.check_call(cmd, stdout=outs, stderr=subprocess.STDOUT)`
`  File "/opt/homebrew/Cellar/python@3.11/3.11.4_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/Users/tpitman/connectedhomeip/.environment/cipd/cipd', 'ensure', '-ensure-file', '/Users/tpitman/connectedhomeip/.environment/cipd/packages.ensure', '-root', '/Users/tpitman/connectedhomeip/.environment/cipd/packages', '-log-level', 'debug', '-json-output', '/Users/tpitman/connectedhomeip/.environment/cipd/packages.json', '-cache-dir', '/Users/tpitman/.cipd-cache-dir', '-max-threads', '0']' returned non-zero exit status 1.`

bzbarsky-apple, 2023-09-08:
> I just tried your clone sequence above and got this error while installing packages:

Yes, like I said "if I fix that problem as in https://github.com/project-chip/connectedhomeip/pull/29132"

tpitman, 2023-09-08:
I decided to start from scratch.

I renamed my esp folder.

I started with the shallow clone method on the esp-idf repo.

When I got to the ./install.sh it gets stuck when checking for esp-idf step.

I have another ticket out on that one in the esp-idf GitHub issues. If you happen to have any ideas on that I would appreciate it.

tpitman, 2023-09-09:
It built this time. Now if I can figure out how to build matter projects from VSCode.

Thanks for the help and tips

bzbarsky-apple, 2023-09-11:
@tpitman Is there something left to do for this issue?

tpitman, 2023-09-11:
I think this issue is done. Closing....

closed 2023-09-11 by tpitman