project-chip/connectedhomeip #29141: [NXP] Rebooting on fail-safe armed during reset is wrong

bzbarsky-apple, 2023-09-08:
### Reproduction steps

`ConfigurationManagerImpl::Init` in src/platform/nxp/mw320/ConfigurationManagerImpl.cpp does this:
```
    if (GetFailSafeArmed(failSafeArmed) == CHIP_NO_ERROR && failSafeArmed)
    {
        ChipLogProgress(DeviceLayer, "Detected fail-safe armed on reboot; initiating factory reset");
        InitiateFactoryReset();
    }
```

But what if the fail-safe was armed for a _second_ commissioning?  This can then wipe out a device that is working just fine.

@doru91 

### Platform

other

### Type

Manually tested with SDK

crlonxp, 2023-10-20:
A PR is created for fixing this issue. 
https://github.com/project-chip/connectedhomeip/pull/29882
Please help to review.
Thank you

crlonxp, 2023-10-23:
The PR is approved and merged. Could this bug be closed?
Thank you.

closed 2023-10-24 by bzbarsky-apple