project-chip/connectedhomeip #29154: [BUG] 4th client device in the same fabric cannot subscribe to the server's attributes

8TRobi8, 2023-09-11:
### Reproduction steps

We have an ESP32-based device, let's call it a "controller" and we have Android phones with our application. The app connects to the controller via Matter and subscribes to all available attributes of all nodes. First, we noticed that the subscriptions are dropped if the 4th app tries to connect to the controller. It can control the controller's nodes, but cannot subscribe to attribute changes.

Therefore, we decreased the number of the nodes and attributes that are available on the controller to determine if maybe the heap is not enough on the ESP32, but with only 1 bridged Lamp node (with OnOff/On-Off attribute and ModeSelect/CurrentMode attribute) the situation is the same.

Reproduction steps:
- We start the controller with the mentioned 1 bridged Lamp node
- Connect with 3 applications
- Connect with the 4th application

**Current behavior:**
App side:
The 4th app cannot subscribe to the attributes, it got the following error: `Will try to resubscribe to 01:0000000000000003 at retry index 7 after 125620ms due to error IM Error 0x000005C8: General error: 0xc8`

Controller side:
I think this is the correct part of the log which is related here (if more logs are required, I can provide them):
D (254101) chip[DMG]: Building Reports for ReadHandler with LastReportGeneration = 8 DirtyGeneration = 0
D (254107) chip[DMG]: <RE> Sending report (payload has 11 bytes)...
I (254113) chip[EM]: <<< [E:47845i M:152206882] (S) Msg TX to 1:0000000000000002 [8CC3] --- Type 0001:05 (IM:ReportData)
I (254123) chip[IN]: (S) Sending msg 152206882 on secure session with LSID: 11124
D (254133) chip[DMG]: IM RH moving to [AwaitingReportResponse]
I (254137) chip[DMG]: Refresh Subscribe Sync Timer with min 1 seconds and max 10 seconds
D (254146) chip[DMG]: <RE> ReportsInFlight = 3 with readHandler 2, RE has no more messages
D (254154) chip[DMG]: All ReadHandler-s are clean, clear GlobalDirtySet
D (254410) chip[EM]: Retransmitting MessageCounter:108806309 on exchange 47843i Send Cnt 4
I (254411) chip[IN]: (S) Sending msg 108806309 on secure session with LSID: 11122
D (254513) chip[EM]: Retransmitting MessageCounter:152206882 on exchange 47845i Send Cnt 1
I (254514) chip[IN]: (S) Sending msg 152206882 on secure session with LSID: 11124
D (254643) chip[EM]: Retransmitting MessageCounter:35699305 on exchange 47844i Send Cnt 4
I (254644) chip[IN]: (S) Sending msg 35699305 on secure session with LSID: 11123
D (254891) chip[EM]: Retransmitting MessageCounter:152206882 on exchange 47845i Send Cnt 2
I (254892) chip[IN]: (S) Sending msg 152206882 on secure session with LSID: 11124
D (255147) chip[DMG]: Unblock report hold after min 1 seconds
D (255463) chip[EM]: Retransmitting MessageCounter:152206882 on exchange 47845i Send Cnt 3
I (255464) chip[IN]: (S) Sending msg 152206882 on secure session with LSID: 11124
E (255966) chip[EM]: Failed to Send CHIP MessageCounter:108806309 on exchange 47843i sendCount: 4 max retries: 4
E (256217) chip[EM]: Failed to Send CHIP MessageCounter:35699305 on exchange 47844i sendCount: 4 max retries: 4
D (256351) chip[EM]: Retransmitting MessageCounter:152206882 on exchange 47845i Send Cnt 4
I (256351) chip[IN]: (S) Sending msg 152206882 on secure session with LSID: 11124
E (257989) chip[EM]: Failed to Send CHIP MessageCounter:152206882 on exchange 47845i sendCount: 4 max retries: 4
D (258421) chip[IN]: SecureSession[0x3ffb8e20]: MarkAsDefunct Type:2 LSID:11122
I (258421) chip[SC]: SecureSession[0x3ffb8e20]: Moving from state 'kActive' --> 'kDefunct'

### Bug prevalence

always

### GitHub hash of the SDK that was being used

4088a77

### Platform

android, esp32

dhrishi, 2023-09-11:
@wqx6 Can you please check this?

bzbarsky-apple, 2023-09-11:
General error 0xC8 is PATHS_EXHAUSTED.

What paths are actually sent in the relevant subscription requests?  Is it using a single wildcard paths, or listing out paths explicitly?

That said, per spec:

> A publisher SHALL ensure that every fabric the node is commissioned into can support at least three Subscribe Interactions to the publisher and that each subscription SHALL support at least 3 attribute/event paths.

so once you are trying to do four subscriptions on a single fabric you're into "device is being asked to do something the spec does not require" territory.  The SDK _does_ try to handle this by default, but there might be bugs in it, and there are some cases (e.g. incoming request asks for more than 3 paths but we don't have enough paths to allocate to it) where incoming requests will just be denied...  So details about what _exact_ subscribe requests are being sent and what exact code the server is running would be helpful here.

bzbarsky-apple, 2023-09-12:
> Every client subscribes with 12 paths.

OK, that's also asking the device for something it doesn't have to suppport, per spec... What's required, per above, is support for 3 subscriptions per fabric, 3 paths per subscription.

But that definitely explains the behavior, assuming this device is configured for the spec minima.  Since 5 fabrics are required as a minimum, that means 45 subscription paths total is a minimum required for a device.  That happens to be enough for three 12-path subscriptions as long as there is only one fabric involved.  But trying to do a fourth one does not fit (48 > 45), and since it's over the spec path limits it just gets rejected instead of evicting one of the existing subscriptions to allow it.

If you control the device, you could configure it to allow more subscription paths.  But all of this is out of "the spec requires a device to allow this" territory and into "devices might try to allow it, best-effort, if you are lucky".

bzbarsky-apple, 2023-09-12:
And @8TRobi8 you mention freeing up more heap, but subscription paths are allocated out of a fixed-size pool, so that the spec minima can be absolutely guaranteed, not from the heap.

closed 2023-09-12 by bzbarsky-apple