project-chip/connectedhomeip #29163: [Feature] (Android) Return/throw an error when an insufficient IPK is provided

tnguyen-alarm, 2023-09-11:
### Feature description

...
I was building out commissioning on Android and noticed that when I provided an insufficient IPK (less than 16 bytes, 4 bytes specifically) via `onNOCChainGeneration()`, commissioning would hang, but no errors would show up in Logcat. It took a while to figure out that the IPK was the cause of the hanging because I was receiving it from the cloud and not generating one myself.

Just a guess, but I would have expected this method to return/throw an error: https://github.com/project-chip/connectedhomeip/blob/b274b9ebe7ea9a89b9347b29b3be23ad2d18347c/src/controller/java/CHIPDeviceController-JNI.cpp#L165

### Platform

android

### Platform Version(s)

1.1.0.1

pgatti86, 2023-11-02:
@tnguyen-alarm how do you fix the problem?

I'm using version 1.2.0.1 but the last log message is setNOCChain()
My IPK is a random generate array of 16 bytes.
Thanks

tnguyen-alarm, 2023-11-02:
@pgatti86 Are you using the same IPK between the commissioner + controller + accessory device? They need to all be the same for CASE to work.

pgatti86, 2023-11-03:
@tnguyen-alarm Thanks for your reply.
I was not using the same ipk! After your suggestion it works. Thanks

yunhanw-google, 2023-11-09:
@tnguyen-alarm 
onNOCChainGeneration has returned the error, you could raise the exception in java side when error is not 0?

do you know which exact line is failing with insufficient IPK?

here?
https://github.com/project-chip/connectedhomeip/blob/b274b9ebe7ea9a89b9347b29b3be23ad2d18347c/src/controller/java/CHIPDeviceController-JNI.cpp#L225

yunhanw-google, 2024-01-14:
@tnguyen-alarm any reply on this? thanks

tnguyen-alarm, 2024-01-17:
@yunhanw-google Sorry for the extremely late reply. I haven't been working in this area of the code for a bit so it hasn't been as high of a priority.

>onNOCChainGeneration has returned the error, you could raise the exception in java side when error is not 0?

Just to confirm, you're suggesting adding a new parameter to `onNOCChainGeneration`  for the return code? If so, that would work. 

>do you know which exact line is failing with insufficient IPK?

I think this is the line that is failing: https://github.com/project-chip/connectedhomeip/blob/b274b9ebe7ea9a89b9347b29b3be23ad2d18347c/src/controller/java/CHIPDeviceController-JNI.cpp#L217. 

>I provided an insufficient IPK (less than 16 bytes, 4 bytes specifically) via onNOCChainGeneration()

In my original example, my guess is that the Java byte array is padded with 0x00's and so it sneaks by the `jByteArrayIpk.byteSpan().size() == sizeof(ipkValue)` check.

yunhanw-google, 2024-01-17:
https://github.com/project-chip/connectedhomeip/pull/31491/files

closed 2024-01-18 by yunhanw-google