project-chip/connectedhomeip #29164: [SVE] [TC-ACL-2.10] Step 10 fails after DUT is rebooted

jrhees-cae, 2023-09-11:
### Reproduction steps

TC-ACL-2.10 fails step 10 after DUT is rebooted in step 9. It appears that this is because TH1 is still attempting to use the the previous CASE session and therefore fails.

If I remove the PICS_SDK_CI_ONLY from the following step in the YAML (the step immediately after the reboot, but before step 10), I can get the test to run successfully:
```
    - label: "TH1 Connects to the device again"
      PICS: PICS_SDK_CI_ONLY
      cluster: "DelayCommands"
      command: "WaitForCommissionee"
      arguments:
          values:
              - name: "nodeId"
                value: nodeId
```

I am testing with a Thread DUT.

### Bug prevalence

Consistent

### GitHub hash of the SDK that was being used

5ccb774e7e5fe7ead3ff85bec1cda31a459c9c37

### Platform

raspi

### Type

Test Improvement

### Testing

YAML tested

bzbarsky-apple, 2023-09-11:
Yeah, I am not sure why that PICS_SDK_CI_ONLY is there.  @manjunath-grl please take a look?

Separately, it looks like a _lot_ of that YAML is PICS_SDK_CI_ONLY including some things that probably shouldn't be.

manjunath-grl, 2023-09-12:
@bzbarsky-apple Yes, for the above step PICS_SDK_CI_ONLY should not exist, we will fix it in the upcoming PR.
          And for the rest of the steps using PICS  `PICS_SDK_CI_ONLY` added because of the issue [768] (https://github.com/ CHIP -Specifications/chip-certification-tool/issues/768), we made the prompt steps for all TH2 steps and for the sample app we made it as automated steps with `PICS_SDK_CI_ONLY`.

manjunath-grl, 2023-09-20:
@jrhees-cae fixed in PR [29179](https://github.com/project-chip/connectedhomeip/pull/29179) please verify it and close the ticket.

jrhees-cae, 2023-09-20:
Verified

closed 2023-09-20 by jrhees-cae