project-chip/connectedhomeip #29177: Level Control Cluster with Bridge-app

HARRYhelenH, 2023-09-12:
### Reproduction steps

In the bridge-app It is given for the on /off light where I need to integrate the Dimmer instead on on/off light.
 To enable the endpoint as level control cluster tried using emberAfLevelControlClusterServerInitCallback() and emberAfLevelControlClusterServerTickCallback() which didn't work out. The write callback is not called and the level cluster is not handled.

The changes made for Dimmer:
[dimmer_changes.txt](https://github.com/project-chip/connectedhomeip/files/12582649/dimmer_changes.txt)
Logs:
[level_bridge.txt](https://github.com/project-chip/connectedhomeip/files/12582650/level_bridge.txt)

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

release/v1.0

### Platform

esp32

### Type

Platform Issue

HARRYhelenH, 2023-11-03:
I have added the cluster definitions as follows:

> DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(levelControlAttrs)
>     DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::ClusterRevision::Id,    INT16U, 2, 0),
>     DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::FeatureMap::Id,       BITMAP32, 4, 0),
>     DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::CurrentLevel::Id,        INT8U, 1, ZAP_ATTRIBUTE_MASK(WRITABLE)),
>     DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::OnLevel::Id,             INT8U, 1, ZAP_ATTRIBUTE_MASK(WRITABLE)),
>     DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::Options::Id,           BITMAP8, 1, 0),
>     DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::MinLevel::Id,            INT8U, 1, 0),
>     DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::MaxLevel::Id,            INT8U, 1, 0),
>     DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::StartUpCurrentLevel::Id, INT8U, 1, 0),
>     DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::RemainingTime::Id,      INT16U, 2, 0),
> DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();
> 
> constexpr CommandId levelControlIncomingCommands[] = {
>     app::Clusters::LevelControl::Commands::MoveToLevel::Id,
>     app::Clusters::LevelControl::Commands::Move::Id,
>     app::Clusters::LevelControl::Commands::Step::Id,
>     app::Clusters::LevelControl::Commands::Stop::Id,
>     app::Clusters::LevelControl::Commands::MoveToLevelWithOnOff::Id,
>     app::Clusters::LevelControl::Commands::MoveWithOnOff::Id,
>     app::Clusters::LevelControl::Commands::StepWithOnOff::Id,
>     app::Clusters::LevelControl::Commands::StopWithOnOff::Id,
>     kInvalidCommandId,
> };
> 
> DECLARE_DYNAMIC_CLUSTER_LIST_BEGIN(bridgedDimmableLightClusters)
> DECLARE_DYNAMIC_CLUSTER(OnOff::Id, onOffAttrs, onOffIncomingCommands, nullptr),
> DECLARE_DYNAMIC_CLUSTER(LevelControl::Id, levelControlAttrs, levelControlIncomingCommands, nullptr),
> DECLARE_DYNAMIC_CLUSTER(Descriptor::Id, descriptorAttrs, nullptr, nullptr),
> DECLARE_DYNAMIC_CLUSTER(BridgedDeviceBasic::Id, bridgedDeviceBasicAttrs, nullptr, nullptr)
> DECLARE_DYNAMIC_CLUSTER_LIST_END;
> 
> DECLARE_DYNAMIC_ENDPOINT(bridgedDimmableLightEndpoint, bridgedDimmableLightClusters);
> 
> DataVersion gDimmableLight1DataVersions[ArraySize(bridgedDimmableLightClusters)];
> 

and called the init callback
emberAfLevelControlClusterServerInitCallback(dev->GetEndpointId());

When I change the levels from alexa the movetolevel callbacck is not called.

bzbarsky-apple, 2023-11-13:
> When I change the levels from alexa the movetolevel callbacck is not called.

Please attach, not paste, logs from the device covering this interaction @HARRYhelenH

HARRYhelenH, 2023-11-14:
@bzbarsky-apple  attaching the log of level implementation.
[level_bridge.txt](https://github.com/project-chip/connectedhomeip/files/13348126/level_bridge.txt)

bzbarsky-apple, 2023-11-16:
@HARRYhelenH That log does not show any invoke interactions (commands being received by the device), which is what I would expect to see if level is being changed.

HARRYhelenH, 2023-11-17:
@HARRYhelenH  I have added the device to alexa and tested, not with the chip-tool. When I change levels in alexa I don't see any callbacks in the logs.

bzbarsky-apple, 2023-11-17:
I mean that I am not seeing any commands to change level in the logs, or any commands at all.

In which case the issue would presumably be on the Alexa side, and you should report it to them....