project-chip/connectedhomeip #29178: Matter client developer profile expired,can't reinstall

zzqzhp, 2023-09-12:
### Reproduction steps

Matter client developer profile expired,when i reinstall it to my iphone, it alert install failed

### Bug prevalence

when it expired

### GitHub hash of the SDK that was being used

the latest

### Platform

darwin, other

### Anything else?

https://down.dingtalk.com/ddmedia/iwEcAqNwbmcDAQTRAdEF0QLHBrDvAzcv2ce4YgTzZWPyJV8AB9IA1K7tCAAJomltCgAL0gAByB0.png_720x720q90.jpg?bizType=IM&auth_bizType=IM&auth_bizEntity=%7B%22cid%22%3A%2213938413%3A778089402%22%2C%22msgId%22%3A%2216848321325755%22%7D&open_id=13938413

zzqzhp, 2023-09-12:
https://down.dingtalk.com/ddmedia/iwEcAqNwbmcDAQTRAdEF0QLHBrDvAzcv2ce4YgTzZWPyJV8AB9IA1K7tCAAJomltCgAL0gAByB0.png_720x720q90.jpg?bizType=IM&auth_bizType=IM&auth_bizEntity=%7B%22cid%22%3A%2213938413%3A778089402%22%2C%22msgId%22%3A%2216848321325755%22%7D&open_id=13938413

zzqzhp, 2023-09-12:
Description file installation failed
The profile "Bluetooth Central MatterClient Developer Mode" is no longer valid.

bzbarsky-apple, 2023-09-12:
@zzqzhp Known issue, being worked on.

mukesh-simform, 2023-09-12:
Facing the same in MacOS as well, while installing profile. "Bluetooth Central Matter Client Developer Mode"

[image]

bzbarsky-apple, 2023-09-12:
Please stop adding "me too" comments.  The profile is expired.  A new profile with a new expiration date needs to be created and published.  I will update here once I am told that has happened.

bzbarsky-apple, 2023-09-13:
Profile at https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/EnableBluetoothCentralMatterClientDeveloperMode.mobileconfig has been updated.

closed 2023-09-13 by bzbarsky-apple

bzbarsky-apple, 2023-09-14:
@zzqzhp Thank you for noticing this so quickly and reporting!  Hopefully this won't happen again...

zzqzhp, 2023-09-14:
[image]
[image]
[image]

The new profile still installation failed

zzqzhp, 2023-09-14:
[image]
[image]

@bzbarsky-apple  The new profile still installation failed

bzbarsky-apple, 2023-09-14:
@zzqzhp I just tested again, on both MacOS and iOS, and the new profile file works for both...  Please make sure that you're testing the new profile file, freshly downloaded from the link above....

zzqzhp, 2023-09-15:
@bzbarsky-apple ,thanks,the new profile file works