project-chip/connectedhomeip #29181: PlatformMgr().LockChipStack() freezes the entire program?

hungdaqq, 2023-09-12:
At some functions, I need to run PlatformMgr().LockChipStack() and PlatformMgr().UnlockChipStack() to prevent the program from getting segmentation faults (core dump). On some devices, the program runs normally, on others, calling the function causes the program to freeze. Do you have any ideas?

bzbarsky-apple, 2023-09-12:
> Do you have any ideas?

If you're calling `LockChipStack()` when the stack is already locked, it will deadlock.

What functions are involved here?  Where are they being called from?  Generally functions should be pretty clear about whether they are running on the Matter event loop (hence with the lock held) or not.

hungdaqq, 2023-09-13:
Thanks for replying so quickly. How do i know a stack is already locked or not? 

> What functions are involved here

bzbarsky-apple, 2023-09-13:
> How do i know a stack is already locked or not?

Generally by structuring your code so that you know what threading context all functions are being called in.

In this case: what is the callstack to HandleWriteOnOffAttribute when it deadlocks?  What is the callstack when it asserts if the lock/unlock are removed?  That will give you an idea of what's going on.  Presumably you want two separate functions:

1. HandleWriteOnOffAttributeInMatterContext that is run when the Matter stack is locked (e.g. when running off the Matter event loop, etc).
2. HandleWriteOnOffAttribute that is run when the Matter stack is not locked, locks the Matter stack, and calls the other function (e.g. if there are explicit HandleWriteOnOffAttribute somewhere in your code that is not running off the Matter event loop).