project-chip/connectedhomeip #29182: [BUG] [TC_STAT_2_1] Step 17C  has incorrect PICS

mhazley, 2023-09-12:
### Reproduction steps

For Thermostat Cluster, attribute 0x0015 (`MinHeatSetpointLimit`) is optional if the `HEAT` feature is on. 

`MinHeatSetpointLimit` PICS is `TSTAT.S.A0015` and the `HEAT` feature PICS is `TSTAT.S.F00`.

Step 17 in test plan reads: 

> TH reads the MinHeatSetpointLimit attribute from the DUT [and] Verify that the DUT responds with an int16 value.

Step 17C has its PICS set to...
```
PICS: TSTAT.S.A0003 && !TSTAT.S.A0015 && !TSTAT.S.A0016 && !TSTAT.S.F05
```
...meaning that when you do not enable this optional attribute, this step runs and the test fails. 

### Bug prevalence

Always when TSTAT.S.A0015 is not enabled

### GitHub hash of the SDK that was being used

ToT

### Platform

ameba, raspi

bzbarsky-apple, 2023-09-12:
@manjunath-grl @cjandhyala

manjunath-grl, 2023-09-20:
@mhazley Can you please confirm its fixed in PR [29179](https://github.com/project-chip/connectedhomeip/pull/29179) and close the ticket.

closed 2023-09-26 by mhazley