project-chip/connectedhomeip #29183: [BUG] [TC_TSTAT_2_2] Step 10a has the wrong PICS

mhazley, 2023-09-12:
### Reproduction steps

Test plan for step 10a reads:

> Set MinHeatSetpointLimit and MaxHeatSetpointLimit back to reasonable values. (f.e. 700 & 3000 == 7°C & 30°C)

`MinHeatSetpointLimit` PICS is `TSTAT.S.A0015`, `MaxHeatSetpointLimit` PICS is `TSTAT.S.A0016`, and the `HEAT` feature PICS is `TSTAT.S.F00`.

The versions of the test steps are below:

```
    - label:
          "Step 10a: Writes (sets back) default value of MinHeatSetpointLimit"
      command: "writeAttribute"
      attribute: "MinHeatSetpointLimit"
      PICS: TSTAT.S.F01 && TSTAT.S.A0015 && TSTAT.S.A0003
      arguments:
          value: AbsMinHeatSetpointLimitValue

    - label:
          "Step 10a: Writes (sets back) default value of MinHeatSetpointLimit"
      command: "writeAttribute"
      attribute: "MinHeatSetpointLimit"
      PICS: TSTAT.S.F01 && TSTAT.S.A0015 && !TSTAT.S.A0003
      arguments:
          value: 700

    - label: "Step 10a: Writes (sets back)default value of MaxHeatSetpointLimit"
      command: "writeAttribute"
      attribute: "MaxHeatSetpointLimit"
      PICS: TSTAT.S.F01 && TSTAT.S.A0016 && !TSTAT.S.F05 && TSTAT.S.A0004
      arguments:
          value: AbsMaxHeatSetpointLimitValue

    - label: "Step 10a: Writes (sets back)default value of MaxHeatSetpointLimit"
      command: "writeAttribute"
      attribute: "MaxHeatSetpointLimit"
      PICS: TSTAT.S.F01 && TSTAT.S.A0016 && !TSTAT.S.F05 && !TSTAT.S.A0004
      arguments:
          value: 3000

    - label:
          "Step 10a: Writes MaxHeatSetpointLimit That meets the deadband of 2.5C"
      command: "writeAttribute"
      attribute: "MaxHeatSetpointLimit"
      PICS: TSTAT.S.F01 && TSTAT.S.A0016 && TSTAT.S.F05
      arguments:
          value: 2950
```

As you can see, all the tests have a PICS of `PICS: TSTAT.S.F01` which is the `COOL` feature. 

This means that if the `COOL` feature is not enabled, these values never get reset and the test case fails in a later step that relies on this. 

### Bug prevalence

Always when `COOL` feature is disabled

### GitHub hash of the SDK that was being used

ToT

### Platform

ameba, raspi

### Platform Version(s)

ToT

bzbarsky-apple, 2023-09-12:
@manjunath-grl @cjandhyala

manjunath-grl, 2023-09-20:
@mhazley Can you please confirm it is fixed in PR [29179](https://github.com/project-chip/connectedhomeip/pull/29179) and close the ticket.

closed 2023-09-26 by mhazley