project-chip/connectedhomeip #29188: [SVE] --paa-trust-store-path option ignored when in interactive mode

jrhees-cae, 2023-09-12:
### Reproduction steps

When attempting to commission a DUT which requires --paa-trust-store-path option, the commissioning will fail if it started in interactive mode, but will commission successfully if run from the linux command line.

It appears that the `--paa-trust-store-path /var/paa-root-certs/` parameter is ignored when using interactive mode.

Example:
1. In interactive mode, the following commissioning will fail:
```
pairing ble-thread 1 hex:0e080000000000010000000300001335060004001fffe0020877328e8ca9d1db480708fd3eb1753b97719d05102fe6aaa72a26ef80bc853ff474f3a732030f4f70656e5468726561642d636566350102cef504104b2d61bf68d5222e913a24ee7bfdff600c0402a0fff8 70246099 4003 --paa-trust-store-path /var/paa-root-certs/
```
However, the same command run from the command-line will work:
```
./chip-tool pairing ble-thread 1 hex:0e080000000000010000000300001335060004001fffe0020877328e8ca9d1db480708fd3eb1753b97719d05102fe6aaa72a26ef80bc853ff474f3a732030f4f70656e5468726561642d636566350102cef504104b2d61bf68d5222e913a24ee7bfdff600c0402a0fff8 70246099 4003 --paa-trust-store-path /var/paa-root-certs/ 
```

### Bug prevalence

Consistent

### GitHub hash of the SDK that was being used

5ccb774e7e5fe7ead3ff85bec1cda31a459c9c37

### Platform

raspi

### Type

Test Improvement

### Testing

Manually tested with SDK

bzbarsky-apple, 2023-09-12:
@jrhees-cae Right now the PAA trust store is created once, at stack startup, in chip-tool.  It can't be changed dynamically.

That might be possible to change, though....  But the workaround is to pass the right `--paa-trust-store-path` when doing `interactive start`, as long as you don't need separate PAA things for the separate fabrics...

jrhees-cae, 2023-09-12:
@bzbarsky-apple Thanks Boris! This lowers the priority of this issue, since this workaround can be used by the TC's affected without changing the overall steps.

kendallgoto, 2025-06-06:
Interested in this ticket since I spent some time chasing it down with some confusion. I don't think its necessary for the PAA trust store to change dynamically, since the need there would be niche, but at least flagging an error when `paa-trust-store-path` is passed into an interactive instance command would be helpful.