project-chip/connectedhomeip #29194: [BUG] keepalive ping timeout - no close frame received

rquidute, 2023-09-12:
### Reproduction steps

It seems Websocket closes abnormally, and TH is observing the following error: `sent 1011 (unexpected error) keepalive ping timeout; no close frame received`.

Run **DUT** `chip-all-clusters-app` and **TH** using this PICS files: [Concentration.measurement.PICS.zip](https://github.com/project-chip/connectedhomeip/files/12590036/Concentration.measurement.PICS.zip)

TH Log: [UI_Test_Run_2023_09_12_15_05_03.log](https://github.com/project-chip/connectedhomeip/files/12590043/UI_Test_Run_2023_09_12_15_05_03.log)

TH Backend Log: [Backend-LOG.txt](https://github.com/project-chip/connectedhomeip/files/12590073/Backend-LOG.txt)

TH Screenshot: 
[image]

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

5ccb774

### Platform

python

### Platform Version(s)

N/A

### Anything else?

Related issue: https://github.com/project-chip/connectedhomeip/issues/28806

antonio-amjr, 2025-04-07:
Any update about this issue?

code-overload42, 2026-10-28:
Hello @rquidute ,

This is issue is blocking us, we also mentionned it here (https://github.com/project-chip/certification-tool/issues/1146).

Is this issue being discussed ?

rquidute, 2026-10-28:
Hi @code-overload42 the #1146 report looks like a different root cause: chip-tool's interactive server stops responding after an outbound WriteRequest, and the keepalive timeout is only the symptom. Please keep further discussion on certification-tool#1146 so it stays in one place. 
If it still reproduces on the 1.7 latest version - TH SVE version (`v2.16-beta3+winter2026`), add the TH and SDK versions and logs  there. Until then, the workaround you found (YAML runs of 4 test cases or fewer, or restarting the backend container between runs) is the best option.