project-chip/connectedhomeip #29201: [SVE] [TC-SC-4.1] Various YAML script issues

jrhees-cae, 2023-09-13:
### Reproduction steps

Following issues with TC-SC-4.1

- Step 6:  instructs the user as follows: "./chip-tool pairing open-commissioning-window 1 1 200 2000 3840". However, this assumes that the longDiscriminator is 3840 in the Test Parameters passed into the script. The user needs to change this command to match the longDiscriminator override. Instructions should be updated to instruct the user to use the same discriminator as the longDiscriminator of the DUT.
- Step 7f: is using the wrong PICS value. Although the step is checking for DEVTYPE subtype, the PICS it is using is for MCORE.SC.VENDOR_SUBTYPE. So, this step will fail if the VENDOR_NAME PICS is true, yet the DUT does not support DEVTYPE.
- Step 9: displays a dialog to the user after the commissioning window has expired. If the DUT supports extended discovery and the user delays in dismissing this dialog, it is possible the DUT will have already timed-out its extended discovery, causing the rest of the test steps to fail.
- Step 10h: For devices with extended discovery, step 10h will fail. Step 10h looks for the _CM subtype, but this step should not be included. The test plan is correct for step 10, and says "- subtype _CM must not be present", yet the YAML script is looking for _CM in step 10h.  This step should either be removed, or (if possible) be rewritten to expect failure (since _CM should not be present during extended discovery.

### Bug prevalence

Consistent

### GitHub hash of the SDK that was being used

5ccb774e7e5fe7ead3ff85bec1cda31a459c9c37

### Platform

raspi

### Type

Test Improvement

### Testing

YAML tested

bzbarsky-apple, 2023-09-13:
@Rajashreekalmane @cjandhyala

manjunath-grl, 2023-09-20:
Similar to issue: https://github.com/CHIP-Specifications/chip-test-scripts/issues/649

FMistrorigo, 2024-07-11:
Hi, 
I have the same issue, the test is searching for the long discriminator 3840 and obviously can't find it.
[image]

The problem is in the test or am I missing something?

PS. the link provided by @manjunath-grl is not working anymore.

duvvurunandagopal7, 2024-11-13:
> * support

under test_parameter update your discriminator and short discriminator for this.

 "test_parameters": {
      "discriminator": 416,
      "shortDiscriminator": 1,
      
      Short discriminator, we can derive from (discriminator in hex >>8)&0x0F