project-chip/connectedhomeip #29203: [BUG] TC-IDM-6.2 step 13 and 14 getting INVALID_ACTION

pankore, 2023-09-13:
### Reproduction steps

After applying the commit https://github.com/project-chip/connectedhomeip/commit/5deb59b6caec86fb40726f2574d4bc2d5423106f

In Step 13 and 14, I am still getting INVALID_ACTION after ACL command.

Tested with v1.2 TE2 all-cluster-app:
[TC-IDM-6.2-step13-14.txt](https://github.com/project-chip/connectedhomeip/files/12594385/TC-IDM-6.2-step13-14.txt)

### Bug prevalence

always 

### GitHub hash of the SDK that was being used

5b4f8004662d00bdb111367fec7d3ea978c23372

### Platform

ameba, other

xshuqun, 2023-09-14:
@bzbarsky-apple From the test plan of step 13: "ACL command giving only access for ACL cluster, So except ACL cluster command if try to send any other command will get status as unsupported access."
So is there a need to change the test plan for step 13? to getting status as INVALID ACCESS?
The outcome should be invalid access or unsupported access?

bzbarsky-apple, 2023-09-14:
The test plan is wrong, and it's wrong because it's not doing a _command_.  It's doing a subscribe.  For a subscribe, the right thing here per spec is INVALID_ACTION.

cjandhyala, 2023-09-14:
@bzbarsky-apple the spec says 'If **both** AttributeRequests and EventRequests are empty' then return INVALID_ACTION' but the scenario in the test plan is 'EventRequests set to path which requires a privilege that is not granted for the cluster in the path'

> If the path is a concrete path: 
> Else if reading the event in the path requires a privilege that is not granted to access the cluster in the path, an [EventStatusIB](https://github.com/project-chip/connectedhomeip/issues/29203#_eventstatusib) SHALL be generated with the UNSUPPORTED_ACCESS Status Code.

bzbarsky-apple, 2023-09-14:
@cjandhyala Please read what I quoted above.  Just to re-emphasize:
> * ***If no error free existent paths remain, then EventRequests are considered empty.***

Apollon77, 2024-06-21:
I completely agree here and also summarized that in https://github.com/project-chip/connectedhomeip/issues/34015

cecille, 2026-11-10:
@j-ororke - can you take a peek at this - looks like your new test is expecting (and getting) unsupported access.

j-ororke, 2026-11-10:
Thanks @cecille, and thanks Boris and Ingo for clarifying the spec and test plan above.

TC_IDM_6_2.py was originally written to match the test plan's expected result, an EventStatusIB with "UNSUPPORTED_ACCESS". As clarified above, that's only reported when the subscription also includes a path the subscribing controller can read. Otherwise, the request returns INVALID_ACTION.

I've updated [#74410](https://github.com/project-chip/connectedhomeip/pull/74410) to split steps 14 and 15 into a and b parts to cover both cases. The test uses a second controller with only View access to BasicInformation, while the TH retains its admin access throughout:

- **Step 14a:** Subscribing to AccessControlEntryChanged alone returns INVALID_ACTION.
- **Step 14b:** Subscribing to AccessControlEntryChanged along with BasicInformation events is accepted, with UNSUPPORTED_ACCESS returned for AccessControlEntryChanged.
- **Step 15a:** A wildcard subscription to AccessControl events alone returns INVALID_ACTION.
- **Step 15b:** A wildcard subscription to all events is accepted, with AccessControlEntryChanged omitted from the report.

#74410 also removes the YAML test, so this should resolve the issue when it merges, if that works for everyone.

I'll also update the test plan PR ([chip-test-plans#6453](https://github.com/CHIP-Specifications/chip-test-plans/pull/6453)) to include steps 14a, 14b, 15a, and 15b with matching text.

cecille, 2026-11-10:
Thanks - sounds like a good addition.