project-chip/connectedhomeip #29205: [ Not a Bug ] - Usecase Understanding with Existing EcoSystem.

smishra-volansys, 2023-09-13:
### Reproduction steps

Hi Team, firstly this is not a bug but more of a Query that I have while developing the Matter controller. 

Let's Say,  we have multiple Devices as Alexa,  Google Assistant, and Apple TV in the same environment. 
All these are paired with multiple Matter-supported sensors such as Motion, Door, and Smart Plug. I want to understand a scenario of Multi-Admin with the above eco-system. 

If I have these devices and controllers, How Can I use the sensors such that I can implement the Multi Admin here? 

Also, in context to this, I have a question,

1 ) Let there be a Device D, Now this Device is connected to 2 fabrics (A and B) using Multi Admin. The Device is controlled by Fabric B and is turned on/off. The question is, when the device is changing its state,  Will it update the Controlling Fabric only? i.e. fabric B for this case, or both the Fabrics (A and B)? 

Thanks in Advance! 

### Bug prevalence

NA

### GitHub hash of the SDK that was being used

v1.1-branch

### Platform

other

### Platform Version(s)

v1.1-branch

### Type

Test Improvement

### Anything else?

This is Just a Query and not a Bug ! Thanks!

bzbarsky-apple, 2023-09-13:
> How Can I use the sensors such that I can implement the Multi Admin here?

What is this really asking?  What does "implement the Multi Admin" mean to you?

> The question is, when the device is changing its state, Will it update the Controlling Fabric only?

Device state, especially physical state, is attached to the device, not the fabric.  State that is per-fabric is explicitly labeled fabric-scoped in the spec.  If a light bulb is turned off by some fabric, all fabrics will see it as off.

smishra-volansys, 2023-09-13:
`What is this really asking? What does "implement the Multi Admin" mean to you?`

Can you Please narrate a scenario in which I can use the Multi Admin Feature of Matter based on the available setup mentioned above ?

bzbarsky-apple, 2023-09-13:
You commission a light bulb into two different fabrics.  Now each fabric can turn the bulb on or off.

If that does not answer your question, you need to very clearly explain your question....

smishra-volansys, 2023-09-14:
Thanks @bzbarsky-apple ! It does answer my question !

bzbarsky-apple, 2023-09-14:
> these two controllers need to be in the same thread network

I don't think so, no.  They just need to be on the same IP network...

> If we have Two Controllers from different Manufacturers (Google Nest and Apple TV), is it possible to configure both of them in the same thread Network ? Is it feasible for the practical implementation?

I believe the answers are yes and yes respectively.  I know Apple TV can definitely join a Thread network created by a different border router; I believe using the THClient APIs.  But I am not 100% sure what the right API calls for this are.

smishra-volansys, 2023-09-15:
Thanks for the information.
It looks like it is considered that every thread-based hub/gateway or border router should join the existing thread network first.
It should create its own thread network only if no thread network is available.

bzbarsky-apple, 2023-09-15:
>  It should create its own thread network only if no thread network is available.

That's ideal, but not necessary.  Matter is designed to work even if there are multiple thread networks involved, as long as there's a single IP network.

smishra-volansys, 2023-10-04:
Thankyou for the response over this thread. We were able to replicate this on our end successfully with complete understanding.

closed 2023-10-04 by smishra-volansys