project-chip/connectedhomeip #29211: [BUG] LWIP_PBUF_FROM_CUSTOM_POOLS doesn't seem to be set up in a reasonable way

bzbarsky-apple, 2023-09-13:
### Reproduction steps

LWIP_PBUF_FROM_CUSTOM_POOLS is used by some of our config headers (as in, its _value_ is used in `ReliableMessageProtocolConfig.h`), but where is it supposed to be defined by consumers?  Some things seem to have it in `lwipopts.h`, but our config headers  don't include `lwipopts.h`, right?

How is this supposed to work?

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

b02c9842e3dedc8f81f16760da5c322affdf27ee

### Platform

other

rojer, 2023-09-21:
`PBUF_POOL_SIZE` is one of LwIP's public options hence it's ok to rely on it being defined. LwIP user code can explicitly `#include <lwipopts.h>`, so that seems like a valid solution (when `CHIP_SYSTEM_CONFIG_USE_LWIP` is set, of course).

bzbarsky-apple, 2023-09-22:
Interestingly, we have other headers using LWIP_PBUF_FROM_CUSTOM_POOLS without obviously including lwipopts.h.  But maybe those are just broken?

For now I am going to condition the use of the LwIP things on CHIP_SYSTEM_CONFIG_USE_LWIP, and then we'll see what happens when we actually try to turn on -Wundef on the LwIP-using platforms.

rojer, 2023-09-22:
lwipopts could be transitively picked up via any other lwip include somewhere in the tree. `#include "lwip/opt.h"` is the first thing any lwip include does, opt.h in turn includes lwipopts.h to pick up user settings.
so actually, small correction: the right thing to do is to `#include "lwip/opt.h"`, to make sure defaults are correctly applied.

closed 2023-09-22 by mergify[bot]