project-chip/connectedhomeip #29216: [BUG] CHIP_CONFIG_NON_POSIX_LONG_OPT needs to be defined when false

bzbarsky-apple, 2023-09-13:
### Reproduction steps

Try to compile with `-Wundef`.

@ATmobica 

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

b02c9842e3dedc8f81f16760da5c322affdf27ee

### Platform

core

ATmobica, 2023-09-14:
@hugueskamba  Could you please take over this issue?

bzbarsky-apple, 2023-09-19:
@ATmobica @hugueskamba this is one of two remaining issues that keep us from turning on `-Wundef`.

bzbarsky-apple, 2023-09-19:
@ATmobica @hugueskamba and I guess the real question is why this was not done as a gn build arg or something with fallback defines in CHIPDeviceConfig.h or something like that....

closed 2023-10-05 by mergify[bot]