project-chip/connectedhomeip #29243: [SVE] TC-TCCM-1.1 unable to detect TCCM.S.A0003

pankore, 2023-09-14:
### Reproduction steps

When disable TCCM.S.A0003, Step 4b reading atrributelist failed

CHIP:TOO AttributeList: 8 entries
--
CHIP:TOO [1]: 0
CHIP:TOO [2]: 1
CHIP:TOO [3]: 2
CHIP:TOO [4]: 65528
CHIP:TOO [5]: 65529
CHIP:TOO [6]: 65531
CHIP:TOO [7]: 65532
CHIP:TOO [8]: 65533
Test Failure: The response ([0, 1, 2, 65528, 65529, 65531, 65532, 65533]) is missing [3].

Test logs:
[UI_Test_Run_2023_09_14_12_52_25.log](https://github.com/project-chip/connectedhomeip/files/12605045/UI_Test_Run_2023_09_14_12_52_25.log)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

5ccb774e7e5fe7ead3ff85bec1cda31a459c9c37

### Platform

ameba

### Type

Test Improvement

### Testing

YAML tested

pankore, 2023-09-19:
@bzbarsky-apple any comments on this issue?
In my case, TCCM.S.F00 and TCCM.S.A0003 is disabled by the TH is still asking for A0003 to be included

bzbarsky-apple, 2023-09-19:
I don't know that there is much to comment on.  The test plan is correct.  The YAML is clearly wrong.  For the `PICS_EVENT_LIST_ENABLED` case it has:
```
              contains: [0, 1, 65528, 65529, 65530, 65531, 65532, 65533]
```
and for the case when EventList is disabled it has:
```
              contains: [0, 1, 2, 3, 65528, 65529, 65531, 65532, 65533]
```

which should not have the "2" and "3" in there.  @manjunath-grl

manjunath-grl, 2023-09-20:
@pankore fixed in PR [29179](https://github.com/project-chip/connectedhomeip/pull/29179) please verify it and close the ticket.