project-chip/connectedhomeip #29249: [Darwin] For iOS16.1-16.4 onNOCChainGenerationComplete crash

robinmo, 2023-09-14:
### Reproduction steps

Xcode15.0 beta4
iPhone14pro / iOS16.3
(with iOS framework matter)

integrate with Matter framework
- commission with full chain
- when perform delegate's onNOCChainGenerationComplete crash
```
// MTRNOCChainIssuer
- (void)onNOCChainGenerationNeeded:(CSRInfo *)csrInfo
                   attestationInfo:(AttestationInfo *)attestationInfo
      onNOCChainGenerationComplete:(MTRNOCChainGenerationCompleteHandler)onNOCChainGenerationComplete
```

log details
```
Sep 14 17:59:49 TPs-iPhone-4 runningboardd(RunningBoard)[31] <Notice>: Invalidating assertion 31-3698-40261 (target:[xpcservice<debug.com.xxxx.aria.commissionRequestHandler([application<com.apple.Home.HomeUIService>:3698])>:3699]) from originator [application<com.apple.Home.HomeUIService>:3698]
Sep 14 17:59:49 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^_\M-" [1694685589601] [3693:2164947] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
Sep 14 17:59:49 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^_\M-" [1694685589617] [3693:2164947] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
Sep 14 17:59:49 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685589617] [3693:2164947] CHIP: [CTL] Starting commissioning discovery over DNS-SD
Sep 14 17:59:49 TPs-iPhone-4 runningboardd(RunningBoard)[31] <Notice>: [xpcservice<debug.com.xxxx.aria.commissionRequestHandler([application<com.apple.Home.HomeUIService>:3698])>:3699] set Memory Limits to Hard Active (80)
Sep 14 17:59:49 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685589867] [3693:2166523] CHIP: [CTL] Discovered device to be commissioned over DNS-SD
Sep 14 17:59:51 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^_\M-" [1694685591718] [3693:2166381] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
Sep 14 17:59:51 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685591757] [3693:2164947] CHIP: [CTL] Performing next commissioning step 'ReadCommissioningInfo'
Sep 14 17:59:51 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685591757] [3693:2164947] CHIP: [CTL] Sending request for commissioning information
Sep 14 17:59:52 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685592011] [3693:2166523] CHIP: [CTL] Successfully finished commissioning step 'ReadCommissioningInfo'
Sep 14 17:59:52 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685592011] [3693:2166523] CHIP: [CTL] Performing next commissioning step 'ArmFailSafe'
Sep 14 17:59:52 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685592687] [3693:2166523] CHIP: [CTL] Successfully finished commissioning step 'ArmFailSafe'
Sep 14 17:59:52 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685592688] [3693:2166523] CHIP: [CTL] Performing next commissioning step 'ConfigRegulatory'
Sep 14 17:59:53 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685593024] [3693:2166777] CHIP: [CTL] Successfully finished commissioning step 'ConfigRegulatory'
Sep 14 17:59:53 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685593024] [3693:2166777] CHIP: [CTL] Performing next commissioning step 'SendPAICertificateRequest'
Sep 14 17:59:53 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685593329] [3693:2166777] CHIP: [CTL] Successfully finished commissioning step 'SendPAICertificateRequest'
Sep 14 17:59:53 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685593330] [3693:2166777] CHIP: [CTL] Performing next commissioning step 'SendDACCertificateRequest'
Sep 14 17:59:53 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685593393] [3693:2166777] CHIP: [CTL] Successfully finished commissioning step 'SendDACCertificateRequest'
Sep 14 17:59:53 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685593393] [3693:2166777] CHIP: [CTL] Performing next commissioning step 'SendAttestationRequest'
Sep 14 17:59:53 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685593742] [3693:2166777] CHIP: [CTL] Successfully finished commissioning step 'SendAttestationRequest'
Sep 14 17:59:53 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685593743] [3693:2166777] CHIP: [CTL] Performing next commissioning step 'AttestationVerification'
Sep 14 17:59:53 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685593762] [3693:2166777] CHIP: [CTL] Successfully finished commissioning step 'AttestationVerification'
Sep 14 17:59:53 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685593762] [3693:2166777] CHIP: [CTL] Performing next commissioning step 'SendOpCertSigningRequest'
Sep 14 17:59:55 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685595141] [3693:2166523] CHIP: [CTL] Successfully finished commissioning step 'SendOpCertSigningRequest'
Sep 14 17:59:55 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685595142] [3693:2166523] CHIP: [CTL] Performing next commissioning step 'ValidateCSR'
Sep 14 17:59:55 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685595157] [3693:2166523] CHIP: [CTL] Successfully finished commissioning step 'ValidateCSR'
Sep 14 17:59:55 TPs-iPhone-4 Deco(Matter)[3693] <Notice>: \M-p\M^_\M^T\M-5 [1694685595157] [3693:2166523] CHIP: [CTL] Performing next commissioning step 'GenerateNOCChain'Sep 14 17:59:56 TPs-iPhone-4 runningboardd(RunningBoard)[31] <Notice>: Acquiring assertion targeting [xpcservice<debug.com.xxxx.aria.commissionRequestHandler([application<com.apple.Home.HomeUIService>:3698])>:3699] from originator [application<com.apple.Home.HomeUIService>:3698] with description <RBSAssertionDescriptor| "com.apple.extension.session" ID:31-3698-40285 target:3699 attributes:[	<RBSLegacyAttribute| requestedReason:ViewService reason:ViewService flags:( AllowIdleSleep PreventTaskSuspend PreventTaskThrottleDown WantsForegroundResourcePriority )>,
Sep 14 17:59:56 TPs-iPhone-4 runningboardd(RunningBoard)[31] <Notice>: Assertion 31-3698-40285 (target:[xpcservice<debug.com.xxxx.aria.commissionRequestHandler([application<com.apple.Home.HomeUIService>:3698])>:3699]) will be created as active
Sep 14 17:59:56 TPs-iPhone-4 runningboardd(RunningBoard)[31] <Notice>: Calculated state for xpcservice<debug.com.xxxx.aria.commissionRequestHandler([application<com.apple.Home.HomeUIService>:3698])>: running-active (role: UserInteractiveNonFocal)

```
snapshot
[image]

### Platform

darwin

### Platform Version(s)

iOS16.3

### Type

Manually tested with SDK, Platform validated

robinmo, 2023-09-14:
@bzbarsky-apple 
cc @sharadb-amazon

bzbarsky-apple, 2023-09-14:
@robinmo So the iOS app is crashing?

The screenshot is not terribly helpful in terms of what the actual crash is, because the relevant windows (the lldb output and log) are tiny.  Care to attach (not paste; attach a file) that output here?

That said, some questions:

1. Why the branching in availability if the two branches are identical?
2. What is `onNOCChainGenerationCompleteError`?  Why is this code always passing in a non-nil error to the callback?
3. Where are noc/icac/rcac/ipk/cat coming from?

robinmo, 2023-09-14:
Off office. Tomorrow I will amend it.

robinmo, 2023-09-15:
@bzbarsky-apple 
# 0.U can see a large scale one by click the snapshot.
# 1.This one codes just test, origin one has no branch, I try to directly run the block code
```
 MTROperationalCertificateChain* chain = [[MTROperationalCertificateChain alloc] initWithOperationalCertificate:operationalCertificate
                                                                                               intermediateCertificate:intermediateCertificate
                                                                                                       rootCertificate:rootCertificate
                                                                                                          adminSubject:adminSubject];
        
        completion ? completion(chain, nil) : nil;
        if (error != nil)
        {
            MTRLog(@"");
            *error = nil;
        }
```
but MTROperationalCertificateChain is not available before ios16.4.
# 2.we generate the rcac, icac, ipk, cat on our server, and in our app generate noc by this
```
+ (MTRCertificateDERBytes _Nullable)makeOperationalCertificate:(id<MTRKeypair>)signingKeypair
                                              signingCertificate:(MTRCertificateDERBytes)signingCertificate
                                            operationalPublicKey:(SecKeyRef)operationalPublicKey
                                                        fabricID:(NSNumber *)fabricID
                                                          nodeID:(NSNumber *)nodeID
                                           caseAuthenticatedTags:(NSArray<NSNumber *> * _Nullable)caseAuthenticatedTags
                                                           error:(NSError * __autoreleasing _Nullable * _Nullable)error
{
    MTRCertificateDERBytes NOC;
    if (@available(iOS MatterSupportVersion16_4, *))
    {
        NOC = [MTRCertificates createOperationalCertificate:signingKeypair
                                         signingCertificate:signingCertificate
                                       operationalPublicKey:operationalPublicKey
                                                   fabricID:fabricID
                                                     nodeID:nodeID
                                      caseAuthenticatedTags:[NSSet setWithArray:caseAuthenticatedTags ?: @[]]
                                                      error:error];
    }
    else if (@available(iOS MatterSupportVersion16_1, *))
    {
        NOC = [MTRCertificates generateOperationalCertificate:signingKeypair
                                           signingCertificate:signingCertificate
                                         operationalPublicKey:operationalPublicKey
                                                     fabricId:fabricID
                                                       nodeId:nodeID
                                        caseAuthenticatedTags:caseAuthenticatedTags
                                                        error:error];
    }
    else
    {
        MTRLog(@"");
    }
    
    return NOC;
}
```

It make sense above ios16.4, no crash for` onNOCChainGenerationComplete`

robinmo, 2023-09-15:
latest found
I replace the startup params init method
```
API_AVAILABLE(ios(16.1))
{
    __auto_type * params = [[MTRDeviceControllerStartupParams alloc] initWithOperationalKeypair:keyPair
                                                                         operationalCertificate:noc
                                                                        intermediateCertificate:icac
                                                                                rootCertificate:rcac
                                                                                            ipk:ipk];
//    params = [[MTRDeviceControllerStartupParams alloc] initWithSigningKeypair:keyPair fabricId:1 ipk:ipk];
//    params.nodeId = nodeID;
    params.vendorId = vendorID;
    MTRLog(@"vendorID: %@", vendorID);
    return params;
}
```

then got the error
```
[4431:2517434] CHIP: [EM] OnMessageReceived failed, err = ../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/messaging/ExchangeMgr.cpp:305: CHIP Error 0x00000070: Unsolicited msg with originator bit clear
2023-09-15 10:33:15.839331+0800 Deco[4431:2517387] Generating operational certificate
```

bzbarsky-apple, 2023-09-15:
> U can see a large scale one by click the snapshot.

I know that.  The issue is not the size of the image, it's the size of the windows in the image, which cut off the logs and other debug output.  And since you scrolled up in the lldb window (or it's just scrolled up for some other reason), it's not showing what was going on right before the crash.

Similarly, I know what VerifyOrReturnError/VerifyOrReturnValue do.  You're giving me all sorts of unrelated info but not the info I actually asked for.

Again: paste, don't attach, the logs.  I am not telepathic and can't read your logs at a distance.

> can you implement one for 16.1-16.4

Those releases are done, so no.

But past that: OK.  So you're using the deprecated API because that's all you can do on 16.3, that makes sense.  

I assume the actual call to the `MTRNOCChainGenerationCompleteHandler` is happening asynchronously, and not under the call to `onNOCChainGenerationNeeded:attestationInfo:onNOCChainGenerationComplete:`?

I also assume this is a crash (application closes), not a hang (application freezes up)?

If either of those assumptions is wrong, please let me know.

robinmo, 2023-09-15:
> I know that. The issue is not the size of the image, it's the size of the windows in the image, which cut off the logs and other debug output. And since you scrolled up in the lldb window (or it's just scrolled up for some other reason), it's not showing what was going on right before the crash.

> 
> Again: paste, don't attach, the logs. I am not telepathic and can't read your logs at a distance.

### full related log:
```

2023-09-15 10:32:24 358 AM(-[ICMatterControllerInfoResponse unpack] line:45) [gener🤪][Debug]:MatterLog origin cat: 18446744062547263489🐣🐣🐣🐣🐣

2023-09-15 10:32:24 359 AM(-[ICACMatter getControllerInfoWithPubKey:]_block_invoke line:93) [gener🤪][Debug]:MatterLog=====🤪IoTCloudClient----getControllerInfo====getControllerInfoHandle success

2023-09-15 10:32:24 359 AM(-[🤪MatterPreFetcher getControllerInfoWithPubKey:]_block_invoke line:128) [gener🤪][Debug]:MatterLog step======getControllerInfoWithPubKey 来源于云✅

2023-09-15 10:32:24 359 AM(-[🤪MatterPreFetcher getControllerInfoWithPubKey:]_block_invoke line:137) [gener🤪][Debug]:MatterLog step======🤪MatterOBPreInfo: {"nodeId":89428794108535296,"fabricId":1,"cat":18446744062547263489}

2023-09-15 10:32:24 360 AM(-[🤪MatterPreFetcher getMatterPreInfoWithPubKey:]_block_invoke line:86) [gener🤪][Debug]:MatterLog step======getMatterPreInfoWithPubKey：成功✅

2023-09-15 10:32:24 362 AM(-[🤪Matter ge🤪reCommissionInfoIntern🤪]_block_invoke_2 line:316) [gener🤪][Debug]:MatterLog step======✅获取controller前置信息成功： {"nodeId":89428794108535296,"fabricId":1,"cat":18446744062547263489}, 此处未获取matterIoGGGodeID为0是正常的

2023-09-15 10:32:24 362 AM(-[🤪Matter checkHandle:msg:line:]_block_invoke line:255) [gener🤪][Debug]:MatterLog step======checkHandle getControllerInfoHandle line 321 end status: 🤪HandleStatusSuccess⛵️⛵️⛵️0x28306d270

2023-09-15 10:32:24 971 AM(-[ICACMatter getIoTInfoWithBatchSize:]_block_invoke_2 line:125) [gener🤪][Debug]:MatterLog step======getIoTInfo====getIoTInfoHandle success

2023-09-15 10:32:25.000856+0800 🤪[4431:2512080] v🤪ue为<ICMatterIoTInfoResponse: 0x281c5ea90> {
    availableNodeIds = [
        "89428794567373312";
        "89428794567602689";
        "89428794568270338";
        "89428794567803395";
        "89428794567528964";
        "89428794567741957";
        "89428794567885318";
        "89428794568122887";
        "89428794568053256";
        "89428794568126985"
    ];
    code = 200;
    iotConfig = <ICTSResponseConfig: 0x281fe2160> {
        body = {
            message = "OK";
            data = {
                availableNodeIds = [
                    "89428794567373312";
                    "89428794567602689";
                    "89428794568270338";
                    "89428794567803395";
                    "89428794567528964";
                    "89428794567741957";
                    "89428794567885318";
                    "89428794568122887";
                    "89428794568053256";
                    "89428794568126985"
                ]
            };
            code = 0
        };
        
2023-09-15 10:32:25 002 AM(-[🤪MatterPreFetcher getMatterIoGGGodeId]_block_invoke line:181) [gener🤪][Debug]:MatterLog step======matterIoGGGodeID: 89428794567373312

2023-09-15 10:32:25 002 AM(-[🤪Matter getMatterIoGGGodeID]_block_invoke line:369) [gener🤪][Debug]:MatterLog step======获取MatterIoGGGodeID信息成功✅：89428794567373312

2023-09-15 10:32:25 002 AM(-[🤪Matter checkHandle:msg:line:]_block_invoke line:255) [gener🤪][Debug]:MatterLog step======checkHandle getMatterIoTHandle line 324 end status: 🤪HandleStatusSuccess⛵️⛵️⛵️0x283076800

2023-09-15 10:32:25.002142+0800 🤪[4431:2512080] v🤪ue为{"nodeId":89428794108535296,"ioGGGodeId":89428794567373312,"fabricId":1,"cat":18446744062547263489}
2023-09-15 10:32:25 002 AM(-[🤪Matter ge🤪reCommissionInfoIntern🤪]_block_invoke line:333) [gener🤪][Debug]:MatterLog step======所有前置信息{"nodeId":89428794108535296,"ioGGGodeId":89428794567373312,"fabricId":1,"cat":18446744062547263489}✳️✳️✳️✳️

2023-09-15 10:32:25 003 AM(-[🤪Matter ge🤪reCommissionInfoIntern🤪]_block_invoke line:336) [gener🤪][Debug]:MatterLog step======获取所有前置信息成功✅

2023-09-15 10:32:25 003 AM(-[🤪Matter checkHandle:msg:line:]_block_invoke line:255) [gener🤪][Debug]:MatterLog step======checkHandle ge🤪reCommissionInfoHandle line 339 end status: 🤪HandleStatusSuccess⛵️⛵️⛵️0x283075630
[... 1052 lines cut ...]
Sep 15 11:18:11 TPs-iPhone-4 locationd[3842] <Notice>: {"msg":"Post Application State Change Notification", "notification":"BackgroundRunning", "pid":4484, "bundleId":"debug.com.tplink.aria.commissionRequestHandler"}
Sep 15 11:18:11 TPs-iPhone-4 locationd[3842] <Notice>: {"msg":"#CLIUA AppMonitor notification", "notification":"BackgroundRunning", "pid":4484, "bundleId":"debug.com.tplink.aria.commissionRequestHandler", "ClientKey":"debug.com.tplink.aria"}
Sep 15 11:18:11 TPs-iPhone-4 locationd[3842] <Notice>: {"msg":"skip erasing #CLIUA for RunningBoard Process State. Does not exists", "bundleId":"debug.com.tplink.aria.commissionRequestHandler"}
Sep 15 11:18:11 TPs-iPhone-4 runningboardd(RunningBoard)[3804] <Notice>: [xpcservice<debug.com.tplink.aria.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4483])>:4484] set Memory Limits to Hard Active (80)
Sep 15 11:18:12 TPs-iPhone-4 runningboardd(RunningBoard)[3804] <Notice>: Removed last relative-start-date-defining assertion for process xpcservice<debug.com.tplink.aria.commissionRequestHandler([application<com.apple.Home.HomeUIService>:4483])>
[... cut ...] 34n    SBMostElevatedStateForProcessID = 1;\134n}"}
Sep 15 11:18:12 TPs-iPhone-4 locationd[3842] <Notice>: {"msg":"Post Application State Change Notification", "notification":"Terminated", "pid":4484, "bundleId":"debug.com.tplink.aria.commissionRequestHandler"}
Sep 15 11:18:12 TPs-iPhone-4 locationd[3842] <Notice>: {"msg":"#CLIUA AppMonitor notification", "notification":"Terminated", "pid":4484, "bundleId":"debug.com.tplink.aria.commissionRequestHandler", "ClientKey":"debug.com.tplink.aria"}
```

bzbarsky-apple, 2023-09-15:
Sorry, I meant to say "attach, don't paste" for logs as above....

>  stuck in onNOCChainGenerationComplete at ios16.3

So a hang?  That would make sense if so.

> I do like this, under the call to onNOCChainGenerationNeeded:attestationInfo:onNOCChainGenerationComplete:

Yeah, on 1.3 that will deadlock because it's got a silly dispatch_sync to call `onNOCChainGenerationNeeded:...`.  Just do the call to onNOCChainGenerationComplete async, and that should fix the hang.

robinmo, 2023-09-19:
> Sorry, I meant to say "attach, don't paste" for logs as above....
>
>
> So a hang? That would make sense if so.
>
>
> Yeah, on 1.3 that will deadlock because it's got a silly dispatch_sync to call [...]

Resolved by make a async call to `onNOCChainGenerationNeeded:...`:tada:

robinmo, 2023-09-19:
Now our app can support from 16.1+ for matter.

closed 2023-09-19 by robinmo