project-chip/connectedhomeip #29278: [SVE] TC-FAN-3.1 write percent-setting value incorrect

pankore, 2023-09-15:
### Reproduction steps

Running TC-FAN-3.1 
step 3a: ./chip-tool fancontrol write percent-setting 30 1 1
step 3b and 3c: reading percent-current and percent-setting will return 100 instead of 30

Tested with linux all-cluster-app:
[TC-FAN-3.1.log](https://github.com/project-chip/connectedhomeip/files/12615006/TC-FAN-3.1.log)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

5ccb774e7e5fe7ead3ff85bec1cda31a459c9c37

### Platform

ameba

### Type

Common Cluster Logic

### Testing

YAML tested, Manually tested with SDK

bzbarsky-apple, 2023-09-15:
Just tried on https://github.com/project-chip/connectedhomeip/commit/5ccb774e7e5fe7ead3ff85bec1cda31a459c9c37 and it also works for me there.  Can you please double-check that you are hitting this, and if you are please attach the server log as well as the client one?

pankore, 2023-09-19:
@bzbarsky-apple it is working well for me too, thank you.

closed 2023-09-19 by pankore