project-chip/connectedhomeip #29282: #29232 makes compile error if CHIP_SYSTEM_CONFIG_PACKETBUFFER_LWIP_PBUF_RAM set to 1

wy-hh, 2023-09-15:
Defining `CHIP_SYSTEM_CONFIG_PACKETBUFFER_LWIP_PBUF_RAM` to 1 makes `CHIP_SYSTEM_PACKETBUFFER_FROM_LWIP_POOL` to 0.
Then `CHIP_SYSTEM_CONFIG_PACKETBUFFER_CAPACITY_MAX`, not used for LwIP-based platform, is required in class PacketBuffer as below. 

```
#if CHIP_SYSTEM_PACKETBUFFER_FROM_LWIP_POOL
    static constexpr uint16_t kMaxSizeWithoutReserve = LWIP_MEM_ALIGN_SIZE(PBUF_POOL_BUFSIZE);
#else
    static constexpr uint16_t kMaxSizeWithoutReserve = CHIP_SYSTEM_CONFIG_PACKETBUFFER_CAPACITY_MAX;
#endif
```

bzbarsky-apple, 2023-09-15:
Is kMaxSizeWithoutReserve not used with lwip?  It sure seems to be used....

What is the right setting of kMaxSizeWithoutReserve when doing CHIP_SYSTEM_CONFIG_PACKETBUFFER_LWIP_PBUF_RAM?

bzbarsky-apple, 2023-09-15:
@kpschoedel can you please take a look?

kpschoedel, 2023-09-15:
Did you previously use `#define CHIP_SYSTEM_CONFIG_PACKETBUFFER_LWIP_PBUF_TYPE PBUF_RAM`?
I think then, because of the error that #29208 called out, `CHIP_SYSTEM_PACKETBUFFER_FROM_LWIP_POOL` was (incorrectly) set to 1, and this was never caught because there's no in-tree build using this configuration.

We would preserve the previous behaviour here by changing that line
```
#if CHIP_SYSTEM_PACKETBUFFER_FROM_LWIP_POOL
```
to
```
#if CHIP_SYSTEM_CONFIG_USE_LWIP
```
and I don't immediately see any other unexpected consequences of #29208.

wy-hh, 2023-09-16:
@kpschoedel Matter currently doesn't fully support `PBUF_RAM` with original LWIP code, so there is no platform enabled in public Matter repo and no auto build detects it.
We use `#define CHIP_SYSTEM_CONFIG_PACKETBUFFER_LWIP_PBUF_TYPE PBUF_RAM` to only pass `pbuf_type` paramter to `pbuf_alloc` in `PacketBufferHandle::New` under our repo for some `PBUF_RAM` experimental developments.

I think changing to `#if CHIP_SYSTEM_CONFIG_USE_LWIP` works for our platform.

closed 2023-09-18 by mergify[bot]

wy-hh, 2023-10-20:
Hi @kpschoedel We have a customized memory pool for PBUF_TYPE PBUF_RAM, then we need some macro definitions as below.
- CHIP_SYSTEM_PACKETBUFFER_LWIP_PBUF_TYPE=PBUF_RAM
- CHIP_SYSTEM_PACKETBUFFER_FROM_LWIP_CUSTOM_POOL = 1

But, now, it seem we can't make it happen.

PR #29325 (2023-09-18): Fix LwIP PBUF_RAM configuration
Missed a case of CHIP_SYSTEM_CONFIG_PACKETBUFFER_LWIP_PBUF_RAM, as there are no in-tree builds using this configuration. (Before #29208, this case had used CHIP_SYSTEM_CONFIG_PACKETBUFFER_LWIP_PBUF_TYPE .)

Fixes #29282
commit 15a65e48 (2023-09-18): Fix LwIP PBUF_RAM configuration (#29325)

Missed a case of CHIP_SYSTEM_CONFIG_PACKETBUFFER_LWIP_PBUF_RAM, as
there are no in-tree builds using this configuration. (This used
CHIP_SYSTEM_CONFIG_PACKETBUFFER_LWIP_PBUF_TYPE before #29208.)

Fixes #29282