project-chip/connectedhomeip #29283: Build chip-tool with Custom arm64 toolchain problem

taigila, 2023-09-15:
### Build issue(s)

Hi everyone,

I'm facing the error while trying to cross-compile chiptool for arm64 hardware running linux using toolchain generated using buildroot.

I am using Ubuntu 22.04 on Virtualbox, with AMD64-based processor

I built openssl and glib2.0 with the cross-compiler at toolchain and put it to Documents/AddedLibrary

steps performed:

```
- gn gen --check --fail-on-unused-args --root=examples/chip-tool out/chip_tool --args='target_cpu="arm64" target_os="linux"' --sysroot="/home/taigila/toolchains/iproc-a72/" --system_libdir="/home/taigila/Documents/AddedLibrary/"

ninja -C out/chip_tool/
``` 

These are the results: 

```
ninja: Entering directory `out/chip_tool/'

[65/644] cc obj/gen/include/platform/L...s/openthread/openthread.introspect.c.o

aarch64-buildroot-linux-gnu-gcc: WARNING: unsafe header/library path used in cross-compilation: '-I/usr/include/gio-unix-2.0'

aarch64-buildroot-linux-gnu-gcc: WARNING: unsafe header/library path used in cross-compilation: '-I/usr/include/libmount'

aarch64-buildroot-linux-gnu-gcc: WARNING: unsafe header/library path used in cross-compilation: '-I/usr/include/blkid'

aarch64-buildroot-linux-gnu-gcc: WARNING: unsafe header/library path used in cross-compilation: '-I/usr/include/glib-2.0'

aarch64-buildroot-linux-gnu-gcc: WARNING: unsafe header/library path used in cross-compilation: '-I/usr/lib/x86_64-linux-gnu/glib-2.0/include'

[85/644] c++ obj/third_party/connected...pal_openssl.CHIPCryptoPALOpenSSL.cpp.o

FAILED: obj/third_party/connectedhomeip/src/crypto/cryptopal_openssl.CHIPCryptoPALOpenSSL.cpp.o 

/home/taigila/toolchains/iproc-a72/bin/aarch64-buildroot-linux-gnu-g++ -MMD -MF obj/third_party/connectedhomeip/src/crypto/cryptopal_openssl.CHIPCryptoPALOpenSSL.cpp.o.d -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -I../../examples/chip-tool/third_party/connectedhomeip/src/include -I../../examples/chip-tool/third_party/connectedhomeip/src -Igen/include -I../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/app-common -I../../examples/chip-tool/third_party/connectedhomeip/examples/chip-tool/include -I../../examples/chip-tool/third_party/connectedhomeip/config/standalone -I../../examples/chip-tool/third_party/connectedhomeip/third_party/nlassert/repo/include -I../../examples/chip-tool/third_party/connectedhomeip/third_party/nlio/repo/include -c ../../examples/chip-tool/third_party/connectedhomeip/src/crypto/CHIPCryptoPALOpenSSL.cpp -o obj/third_party/connectedhomeip/src/crypto/cryptopal_openssl.CHIPCryptoPALOpenSSL.cpp.o

../../examples/chip-tool/third_party/connectedhomeip/src/crypto/CHIPCryptoPALOpenSSL.cpp:31:24: fatal error: openssl/bn.h: No such file or directory

compilation terminated.

[93/644] c++ obj/third_party/connected...ibClusterObjects.cluster-objects.cpp.o

FAILED: obj/third_party/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/libClusterObjects.cluster-objects.cpp.o 

/home/taigila/toolchains/iproc-a72/bin/aarch64-buildroot-linux-gnu-g++ -MMD -MF obj/third_party/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/libClusterObjects.cluster-objects.cpp.o.d -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -I../../examples/chip-tool/third_party/connectedhomeip/src/include -I../../examples/chip-tool/third_party/connectedhomeip/src -Igen/include -I../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/app-common -I../../examples/chip-tool/third_party/connectedhomeip/examples/chip-tool/include -I../../examples/chip-tool/third_party/connectedhomeip/config/standalone -I../../examples/chip-tool/third_party/connectedhomeip/third_party/nlassert/repo/include -I../../examples/chip-tool/third_party/connectedhomeip/third_party/nlio/repo/include -c ../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/cluster-objects.cpp -o obj/third_party/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/libClusterObjects.cluster-objects.cpp.o

In file included from ../../examples/chip-tool/third_party/connectedhomeip/src/access/SubjectDescriptor.h:23:0,

                 from ../../examples/chip-tool/third_party/connectedhomeip/src/app/EventLoggingTypes.h:20,

                 from ../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/cluster-objects.h:28,

                 from ../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/cluster-objects.cpp:20:

../../examples/chip-tool/third_party/connectedhomeip/src/lib/core/CASEAuthTag.h:66:84: error: array must be initialized with a brace-enclosed initializer

     std::array<CASEAuthTag, kMaxSubjectCATAttributeCount> values = { kUndefinedCAT };

                                                                                    ^

cc1plus: error: unrecognized command line option ‘-Wno-cast-function-type’ [-Werror]

cc1plus: error: unrecognized command line option ‘-Wno-unknown-warning-option’ [-Werror]

cc1plus: all warnings being treated as errors

[94/644] ACTION //third_party/connecte...build/toolchain/linux:linux_arm64_gcc)

ninja: build stopped: subcommand failed.

``` 
I don't understand why ninja don't look for libraries in "/home/taigila/Documents/AddedLibrary/", instead they look for it in my ubuntu root. Also, eventhough i have a built openssl (from the source code), why are they still looking for 'openssl/bn.h"?

Thank you.

### Platform

other

taigila, 2023-09-22:
@andy31415 @qq8512852 hi sirs, i was looking at cross-compiling issues, and it seems that you both have some experience with cross-compilation with Matter. I am now trying to compile Chip-tool using a buildroot-based arm64 toolchain. I have compiled openssl and glib2.0 with my toolchain. I wonder if any of you would be guide and help me on how you guys did cross-compilation for chip-tool before? especially how to direct GN to my compiled library? 

Thank you.

andy31415, 2023-09-26:
https://github.com/project-chip/connectedhomeip/blob/master/integrations/docker/images/stage-1/chip-build-crosscompile/README.md has some notes about building a sysroot that supports cross-compilation.

https://github.com/project-chip/connectedhomeip/blob/master/scripts/build/builders/host.py#L394 shows some flags that are set when cross compiling. In particular target_cpu and sysroot has to be set (which you seem to already do) however I am unsure about the other arguments.

Based on the error I wonder if this is some compiler version difference. 

Overall I cannot provide support for this - the matter SDK is complex as is and driven by volunteers and we are unable to generally support possible variations and build environments. All I can say is that chip-tool can be currently cross-compiled using the arm64 sysroot and is tested in CI, however your environment is likely different.

qq8512852, 2023-10-06:
> @andy31415 @qq8512852 hi sirs, i was looking at cross-compiling issues, and it seems that you both have some experience with cross-compilation with Matter. I am now trying to compile Chip-tool using [...]

Sorry, I compiled it last year.  I have already forgotten the details.

taigila, 2023-10-11:
hi @andy31415 @qq8512852 thank you so much for the answers. I will look into it. Thank you.

taigila, 2023-10-30:
Found the problem, and it was because of my GCC version. Thank you

closed 2023-10-30 by taigila

DNSalaf, 2024-06-30:
hi @taigila, I am facing a similar issue... what GCC version have you used?