project-chip/connectedhomeip #29284: [CERT-TEST-FAILURE] TestFabricRemovalWhileSubscribed invalid in step 7

Apollon77, 2023-09-15:
### Feature Area

Other

### Test Case

TestFabricRemovalWhileSubscribed

### Reproduction steps

The test is deleting it's own fabric is step 7, but then also expects to receive the Removal response which is especially described in the specs (11.17.6.12.) like:

> 3. If the FabricIndex equals the accessing fabric index, then the device SHALL begin the process of irrevocably deleting all associated Fabric-Scoped data, including Access Control Entries, bind­ ings, group keys, operational certificates, etc. Any remaining Trusted Roots no longer refer­ enced by any operational certificate SHALL also be removed. All secure sessions, exchanges and interaction model constructs related to the Operational Identity under the given Fabric SHALL also be removed. **Since this operation involves the removal of the secure session data that may underpin the current set of exchanges, the Node invoking the command SHOULD NOT expect a response before terminating its secure session with the target.**

### Bug prevalence

reproducable anytime

### GitHub hash of the SDK that was being used

current master

### Platform

other

### Anything else?

```
[1694762175469] [14219:6944507] [TOO]  ***** Test Step 7 : Remove single own fabric

[1694762175469] [14219:6944507] [DMG] ICR moving to [AddingComm]
[1694762175469] [14219:6944507] [DMG] ICR moving to [AddedComma]
[1694762175469] [14219:6944507] [EM] <<< [E:64703i S:31700 M:163041026] (S) Msg TX to 2:0000000012344321 [7A5E] --- Type 0001:08 (IM:InvokeCommandRequest)
[1694762175469] [14219:6944507] [IN] (S) Sending msg 163041026 on secure session with LSID: 31700
[1694762175469] [14219:6944507] [DMG] ICR moving to [CommandSen]
[1694762175469] [14219:6944507] [DL] _OnPlatformEvent
[1694762175474] [14219:6944507] [EM] >>> [E:51463r S:31700 M:3984090030] (S) Msg RX from 2:0000000012344321 [7A5E] --- Type 0000:40 (SecureChannel:StatusReport)
[1694762175474] [14219:6944507] [EM] Handling via exchange: 51463r, Delegate: 0x6000039ccda0
[1694762175474] [14219:6944507] [EM] Received CloseSession status message, closing session
[1694762175474] [14219:6944507] [IN] SecureSession[0x6000007e0200]: MarkForEviction Type:2 LSID:31700
[1694762175474] [14219:6944507] [SC] SecureSession[0x6000007e0200, LSID:31700]: State change 'kActive' --> 'kPendingEviction'
[1694762175474] [14219:6944507] [DMG] Time out! failed to receive invoke command response from Exchange: 64703i
[1694762175474] [14219:6944507] [TOO]  ***** Test Failure: status value mismatch: expected 0 but got 1

[1694762175474] [14219:6944507] [DMG] ICR moving to [AwaitingDe]
[1694762175474] [14219:6944507] [IN] SecureSession[0x6000007e0200]: Released - Type:2 LSID:31700
[1694762175474] [14219:6944507] [DL] _OnPlatformEvent
[1694762175474] [14219:6944506] [CTL] Shutting down the commissioner
[1694762175474] [14219:6944506] [CTL] Stopping commissioning discovery over DNS-SD
[1694762175474] [14219:6944506] [CTL] Shutting down the controller
[1694762175474] [14219:6944506] [IN] Expiring all sessions for fabric 0x2!!
[1694762175474] [14219:6944506] [FP] Forgetting fabric 0x2
[1694762175474] [14219:6944506] [TS] Pending Last Known Good Time: 2023-08-14T23:38:54
[1694762175474] [14219:6944506] [TS] Previous Last Known Good Time: 2023-08-14T23:38:54
[1694762175474] [14219:6944506] [TS] Reverted Last Known Good Time to previous value
```

As you can see the device is sending a sesion close becuse the secure session data are removed as defined. But the test step fails because it did not received the response of the removeFabric call.

The matter.js log:
```
2023-09-15 09:16:15.471 DEBUG MessageExchange New exchange protocol: 1 id: 64703 active retransmit ms: 300 idle retransmit ms: 5000 retries: 5
2023-09-15 09:16:15.471 DEBUG MessageExchange Message « id: 63875/64703/163041026 type: 1/8 reqAck: true payload: 1528002801360215370024000024013e24020a18350124000118181824ff0a18
2023-09-15 09:16:15.471 DEBUG InteractionServer Received invoke request from udp://fe80::419:56de:f7c4:1373%en7:54532 on session secure/63875: MA-rootdevice(0x0)/OperationalCredentials(0x3e)/removeFabric(0xa), suppressResponse=false
2023-09-15 09:16:15.471 DEBUG CommandServer Invoke removeFabric with data {"fabricIndex":1}
2023-09-15 09:16:15.471 DEBUG EventHandler Received event: {"eventNumber":3,"eventId":2,"clusterId":40,"endpointId":0,"epochTimestamp":1694762175471,"priority":1,"data":{"fabricIndex":1}}
2023-09-15 09:16:15.472 DEBUG MatterDevice Remove CASE session secure/63875
2023-09-15 09:16:15.472 DEBUG MessageExchange New exchange protocol: 0 id: 51463 active retransmit ms: 300 idle retransmit ms: 5000 retries: 5
2023-09-15 09:16:15.472 DEBUG ExchangeManager Message » id: 31700/51463/3984090030 type: 0/64 reqAck: false payload: 0000000000000300
2023-09-15 09:16:15.498 DEBUG CommandServer Invoke removeFabric response : {"statusCode":0,"fabricIndex":1}
2023-09-15 09:16:15.499 DEBUG ExchangeManager Message » id: 63875/64703/3984090031 type: 1/9 acked: 163041026 reqAck: true payload: 1528003601153500370024000024013e2402081835012400002401011818181824ff0a18
```

So is this wrong in the 1.1/2 Matter specs, or does the test have invalid expectations here?

Apollon77, 2023-09-21:
@bzbarsky-apple sorry to tag you here directly but with your spec knowledge it would be great to know if we have a spec issue here too or if the test assumptions should be adjusted. In fact i could also have opened a chip bug because the removeFabric fails on Chip level which it should not because it „shall“ Be expected to get no response ;-)

bzbarsky-apple, 2023-09-21:
@Apollon77 The basic spec situation is:

1. The server MAY send a response before tearing things down.
2. The client SHOULD NOT expect this response.

The SDK situation is that the SDK goes to some pains to in fact send the response.  `TestFabricRemovalWhileSubscribed` and `TestSelfFabricRemoval` are fine, since they are SDK self-tests.  `TestFabricRemovalWhileSubscribed` in particular is a test that verifies the _server_ does not crash in this situation; it's a regression test for https://github.com/project-chip/connectedhomeip/issues/20678

The cert tests, on the other hand, need to get fixed.  @cjandhyala @manjunath-grl I filed https://github.com/CHIP-Specifications/chip-test-scripts/issues/663 on that.

bzbarsky-apple, 2023-09-21:
Looks like `Test_TC_CADMIN_1_6` has the same issue.

Apollon77, 2023-09-21:
Yes I stopped some time to add more cases. Can check later if I have more on my list I stumbled over. And thank you again for the clarification. 

I might consider now for matter.js to change our implementation to match with Chip (destroy Session a bit later to send response out in any case) because else we can not be sure that clients/controller really respect the "should not" and because of missing formal controller certification and validation this is else completely unsure how they react to devices doing it differently then Chip.

Apollon77, 2023-09-21:
> Looks like Test_TC_CADMIN_1_6 has the same issue.

I updated the list in first post. these are the test cases I'm aware of.

Apollon77, 2023-09-23:
@bzbarsky-apple Discussed all this I have one (a bit) off-topic question ... How can I address that Apple in most iOS 16.x and also now 17.x do not remove all fabrics when a device gets deleted? I think to remember it was correct in iOS 16.1 or such, but then broke somewhen. Furthermore it seems to try to contact the device after having removed that fabric with a destination id of the removed fabric ... In fact this controller behaviour is turning the device "Unusable" because it is not "uncommissioned" afterwards (sorry to say, but at least Google does all right).
I thought/hoped that it is because of the fact that we (matter.js) were killing the sessions immediatly on removal (like discussed above),  but also now that I changed the session eviction to be a bit "delayed" as chip too does, the behaviour did not really changed.

Here a log from device side:
* During commissioning apple created two fabrics (index 1 and 2)

```
2023-09-23 15:37:04.094 INFO  CaseServer           session 5225 resumed with udp://fe80::1c1c:4eaf:c9ff:472a%en0:60350 for Fabric 000000000D625E26(index 1) and PeerNode 27951A9BAF2C0627

2023-09-23 15:38:05.108 DEBUG MessageExchange      New exchange protocol: 1 id: 30648 active retransmit ms: 300 idle retransmit ms: 5000 retries: 5
2023-09-23 15:38:05.108 DEBUG MessageExchange      Message « id: 5225/30648/38128139 type: 1/2 reqAck: true payload: 1536001724020024031d2404031818290324ff0118
2023-09-23 15:38:05.109 DEBUG InteractionServer    Received read request from udp://fe80::1c1c:4eaf:c9ff:472a%en0:60350 on session secure/5225: attributes:MA-rootdevice(0x0)/Descriptor(0x1d)/partsList(0x3), events:none isFabricFiltered=true
2023-09-23 15:38:05.109 DEBUG InteractionServer    Read attribute from udp://fe80::1c1c:4eaf:c9ff:472a%en0:60350 on session secure/5225: MA-rootdevice(0x0)/Descriptor(0x1d)/partsList(0x3)=[1] (version=1965097353)
2023-09-23 15:38:05.110 DEBUG InteractionMessenger Sending DataReport chunk with 1 attributes and 0 events: 38 bytes
2023-09-23 15:38:05.110 DEBUG InteractionMessenger checkDecode {"attributeReports":[{"attributeData":{"dataVersion":1965097353,"path":{"endpointId":0,"clusterId":29,"attributeId":3},"data":[{"typeLength":{"type":22}},{"typeLength":{"type":4,"length":0},"value":1},{"typeLength":{"type":24}}]}}],"suppressResponse":false,"interactionModelRevision":10}
2023-09-23 15:38:05.110 DEBUG ExchangeManager      Message » id: 5225/30648/185973258 type: 1/5 acked: 38128139 reqAck: true payload: 153601153501260089012175370124020024031d240403183602040118181818280424ff0a18
2023-09-23 15:38:05.119 DEBUG MessageExchange      Message « id: 5225/30648/38128140 type: 1/1 acked: 185973258 reqAck: true payload: 1524000024ff0118
2023-09-23 15:38:05.119 DEBUG ExchangeManager      Message » id: 5225/30648/185973259 type: 0/16 acked: 38128140 reqAck: false payload: 
2023-09-23 15:38:05.123 DEBUG MessageExchange      New exchange protocol: 1 id: 30649 active retransmit ms: 300 idle retransmit ms: 5000 retries: 5
2023-09-23 15:38:05.123 DEBUG MessageExchange      Message « id: 5225/30649/38128141 type: 1/2 reqAck: true payload: 1536001724020024031d2404011818290324ff0118
2023-09-23 15:38:05.123 DEBUG InteractionServer    Received read request from udp://fe80::1c1c:4eaf:c9ff:472a%en0:60350 on session secure/5225: attributes:MA-rootdevice(0x0)/Descriptor(0x1d)/serverList(0x1), events:none isFabricFiltered=true
2023-09-23 15:38:05.123 DEBUG InteractionServer    Read attribute from udp://fe80::1c1c:4eaf:c9ff:472a%en0:60350 on session secure/5225: MA-rootdevice(0x0)/Descriptor(0x1d)/serverList(0x1)=[29,31,40,48,49,51,60,62,63] (version=1965097353)
2023-09-23 15:38:05.124 DEBUG InteractionMessenger Sending DataReport chunk with 1 attributes and 0 events: 54 bytes
2023-09-23 15:38:05.124 DEBUG InteractionMessenger checkDecode {"attributeReports":[{"attributeData":{"dataVersion":1965097353,"path":{"endpointId":0,"clusterId":29,"attributeId":1},"data":[{"typeLength":{"type":22}},{"typeLength":{"type":4,"length":0},"value":29},{"typeLength":{"type":4,"length":0},"value":31},{"typeLength":{"type":4,"length":0},"value":40},{"typeLength":{"type":4,"length":0},"value":48},{"typeLength":{"type":4,"length":0},"value":49},{"typeLength":{"type":4,"length":0},"value":51},{"typeLength":{"type":4,"length":0},"value":60},{"typeLength":{"type":4,"length":0},"value":62},{"typeLength":{"type":4,"length":0},"value":63},{"typeLength":{"type":24}}]}}],"suppressResponse":false,"interactionModelRevision":10}
2023-09-23 15:38:05.124 DEBUG ExchangeManager      Message » id: 5225/30649/185973260 type: 1/5 acked: 38128141 reqAck: true payload: 153601153501260089012175370124020024031d240401183602041d041f0428043004310433043c043e043f18181818280424ff0a18
2023-09-23 15:38:05.125 DEBUG MessageExchange      New exchange protocol: 1 id: 30650 active retransmit ms: 300 idle retransmit ms: 5000 retries: 5
2023-09-23 15:38:05.125 DEBUG MessageExchange      Message « id: 5225/30650/38128142 type: 1/2 reqAck: true payload: 1536001724020124031d2404011818290324ff0118
2023-09-23 15:38:05.125 DEBUG InteractionServer    Received read request from udp://fe80::1c1c:4eaf:c9ff:472a%en0:60350 on session secure/5225: attributes:MA-onoffpluginunit(0x1)/Descriptor(0x1d)/serverList(0x1), events:none isFabricFiltered=true
2023-09-23 15:38:05.125 DEBUG InteractionServer    Read attribute from udp://fe80::1c1c:4eaf:c9ff:472a%en0:60350 on session secure/5225: MA-onoffpluginunit(0x1)/Descriptor(0x1d)/serverList(0x1)=[3,4,5,6,29 [... cut ...]
2023-09-23 15:39:05.335 DEBUG MdnsBroadcaster      Announcement: Fabric id: 415cba0dfe49d4ea/324348170 qname: 415CBA0DFE49D4EA-000000001355290A._matter._tcp.local port: 5541 interface: en7
2023-09-23 15:39:05.336 INFO  UdpMulticastServer   en5: send EHOSTUNREACH ff02::fb:5353
2023-09-23 15:39:18.619 DEBUG MdnsBroadcaster      Announcement: Fabric id: 415cba0dfe49d4ea/324348170 qname: 415CBA0DFE49D4EA-000000001355290A._matter._tcp.local port: 5541 interface: lo0
^C2023-09-23 15:39:19.110 DEBUG EventHandler         Received event: {"eventNumber":7,"eventId":1,"clusterId":40,"endpointId":0,"epochTimestamp":1695476359110,"priority":2}
```

bzbarsky-apple, 2023-09-23:
> when a device gets deleted

Deleted from where?  When you commission a device into Apple Home, it gets added to the Apple Home fabric and the System Commisioner fabric.  The latter is used to open commissioning windows for any other fabrics that want to commission the device, including if the device is removed from the Apple Home fabric.

There is separate UI, as I understand, for removing devices entirely from the System Commissioner fabric.  It's not part of the Home UI, since it's not attached to Home.

> Furthermore it seems to try to contact the device after having removed that fabric with a destination id of the removed fabric

I don't know why Home is doing that.  Please file a feedback assistant ticket and let me know the id and I will pass it along to the right people.  See https://github.com/project-chip/connectedhomeip/issues/28256#issuecomment-1703123417 for an example set of instructions for filing the feedback assistant ticket.

Apollon77, 2023-09-24:
In fact I just used Apple Home to add a device (chip tool or also matter.js as seen above) and 1 minute later removed this device again via Apple home. So i was not doing anything else or any other secondary commissioning or such. Do the "pure basics".

I will try to open such a feedback ticket.

Apollon77, 2023-09-24:
@bzbarsky-apple Ok ... with the knowledge that the second faric is the system commisioner fabric which can be seen and managed via iOS Settings - General - Matter devices then yes this is correct and with "remove from keyring" (no idea how it is called in english, my ios is german, so I translated) you can remove the second fabric and then the devcie gets back to the right state.

If this makes sense if a device was onlycommissioned via Apple Home and so the user added the devcie in Apple Home and expects also a deletion to only handled there is another story ;-) So that might be a user feedback ...

As I tried delete today also the hub did not tried to contact the devcie after removal ... so that what was clearly reproducable yesterday did not happen :-)

bzbarsky-apple, 2023-09-28:
> If this makes sense if a device was onlycommissioned via Apple Home and so the user added the devcie in Apple Home 

Fwiw, it's true no matter what application commissions the device.  It's commissioned into the System Commissioner fabric and into the fabric of the application doing the commissioning.  After that the "Matter devices" UI can be used to open new commissioning windows to commission into new fabrics, etc, as I understand it.

Apollon77, 2023-09-28:
I agree from a tech perspective, my main comments were from a pure users perspective: When I add the devcie via Apple Home and then remove it in Apple Home too (so I never got any contact with any "System commissioner UI" or even know it exists) then I would expect that I can commission the device to e.g Amazon or Google or such again directly ... but thats not possible. So you need to do a factory reset to remove the system commissioner fabric on the device (but the Apple System comissionier still has that entry then and it might not go away ever).