project-chip/connectedhomeip #29296: [BUG] WHy chip tools still send InteractionModelRevision = 1 instead of 10?

Apollon77, 2023-09-15:
### Reproduction steps

When you look at the commands that chip tool and also the chip apps send all interaction messages are send with 

InteractionModelRevision = 1

The Matter spec 1.0 had set this revision number to 10, so why this is not used? versions 1..9 are some "versions of the Zigbee Cluster Library chapter 2 which combined the interac­ tion model with encoding" ...

### Bug prevalence

always

### GitHub hash of the SDK that was being used

any

### Platform

other

bzbarsky-apple, 2023-09-16:
@Apollon77 Please list the actual github hash involved, not "any".   This was fixed in https://github.com/project-chip/connectedhomeip/pull/28645 over a month ago.

bzbarsky-apple, 2023-09-16:
@Apollon77 That said, if you are in fact seeing that on tip, please reopen as needed.  Also, note that the spec has been updated to make it clear that the revisions the 1.0 SDK sent did not match the 1.0 spec.  How things got there... that's a separate story.

closed 2023-09-16 by bzbarsky-apple

Apollon77, 2023-09-16:
@bzbarsky-apple sorry for it ... I thought I updated but in fact did not ... basically I run (again ... had that before, learned and forgot ;-) ) into https://github.com/project-chip/connectedhomeip/pull/29299 ... so now I tried to update and run into https://github.com/project-chip/connectedhomeip/issues/29300 because newest builds of chip tool do not have tests in any more and no docs were updated :-(

Apollon77, 2023-09-16:
But yes now after I managed to somehow get it running the reported revision is 11 (most likely Matter 1.2 coming .. soooo) is fine now. Thank you