project-chip/connectedhomeip #29300: [Tooling] current chip tool builds do not have tests included

Apollon77, 2023-09-16:
### Reproduction steps / Feature

In "former times" chip-tool builds using this repo allowed to run the tests using "chip-tool test". last version from August did that still. I now updated my repo and rebuild the binaries and now the "tests" subcommand is no longer availanble in chip-tool builds.

I also did not found any info in Readmes or docs on how to build it with the tests.

What do I need to do?

### Platform

darwin

### Platform Version(s)

master

### Type

Manually tested with SDK

Apollon77, 2023-09-16:
Ok, I think I found the reason for that move ... https://github.com/project-chip/connectedhomeip/pull/28783

Then this issue is about to update pages like https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/chip_tool_guide.md#running-a-test-suite-against-a-paired-peer-device and such that all point to the no longer working way.

woody-apple, 2023-10-19:
@cjandhyala or @raju-apple could you take a look at updating the documentation here?

woody-apple, 2025-01-29:
This bug is 2 years old, please reopen if this still is an issue.

closed 2025-01-29 by woody-apple