project-chip/connectedhomeip #29302: [BUG] [DARWIN] MTRSetupPayload doesn't parse the same thing

qawitherev, 2023-09-18:
### Reproduction steps

I want create MTRSetupPayload using both qrCode (MT:xxxxx) and using manual pairing key (11-digit key) using: 

`MTRSetupPaylad(onboardingPayload: theString)`

However, upon further inspecting, the same payload doesn't produce the same `MTRSetupPayload`. 

From the same payload: 

Using QRCode : 
`JSONMap is {manualEntryCode: 34970112332, productId: 32768, discriminator: 3840}`

Using manual pairing code: 
`JSONMap is {manualEntryCode: 34970112332, discriminator: 15, productId: 0}`

I also tested with the deprecated API i.e. `MTRManualSetupPayloadParser` to create the MTRSetupPayload, but still getting the same result. 

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

-

### Platform

darwin

### Anything else?

Correct me if I am wrong, 

1. QRPayload and its manual pairing key should produce the same `MTRSetupPayload`

bzbarsky-apple, 2023-09-18:
A Matter QR code encodes a "long discriminator" which is 12 bits.  A Matter manual code only encodes a "short discriminator", which is the top 4 bits of the 12-bit code.  You can see that here: `3840 == 0xf00` and `15 = 0xf`.  Given an MTRSetupPayload you can examine `hasShortDiscriminator` to see what its `discriminator` value means (which will depend on what sort of setup code it came from).

Further, a Matter QR code always has a vendor id and product id, but a Matter manual code only has those when it's a 21-digit code; 11-digit ones do not encode a vendor id or product id.  You also see that above, with `productId: 0`.

The only thing that is guaranteed to be the same across the two payload types is the `setupPasscode`, which is the one thing that you absolutely have to have to commission a Matter device.  All the other information in there is just for filtering which devices you might try to discover and commission.

closed 2023-09-18 by bzbarsky-apple

qawitherev, 2023-09-18:
qrcode: long discriminator (12bit)
11 digit code: short discriminator (first 4bit), no vendor and product id
12 digit code: short/long discriminator?, has vendor and product id 

always the same both payload type: setupPasscode, also the most important thing that identifies the device, the vendor, product id is just to further filter it