project-chip/connectedhomeip #29309: [BUG]

EIC-deval-jariwala, 2023-09-18:
### Reproduction steps

Hi. I have Matter thread enabled Plug. I am using Google Nest (2nd Gen) as a thread border router and am able to commission and operate it successfully using the Google Home app.

Build using the below build argument: (for both x86 and arm machine)
--args='is_debug=false optimize_for_size=true symbol_level=1 target_os="linux" '

I use pairing code shared via Google app and commission it as second admin in host x86 machine. It was able to commission and operate using the below command.

$ ./chip-tool pairing code 8 13787619912 --commissioner-name beta --paa-trust-store-path connectedhomeip/credentials/production/paa-root-certs.

But after another iteration. i.e. reset the device, pair using the Google Home app, and commission using 2nd admin. via my custom Linux device using the same command which is on the same wifi network. But it was not able to commission and gives err.
CHIP:TOO: Run command failure: ../../third_party/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:255: CHIP Error 0x00000032: Timeout 

It fails after the below request after multiple retries.
-> Sent PBKDF param request 
-> PASESession timed out while waiting for a response from the peer. Expected message type was 3

Please find attached complete err log. 
[thread_with_second_admin_notok_with_google_nest.txt](https://github.com/project-chip/connectedhomeip/files/12645551/thread_with_second_admin_notok_with_google_nest.txt)

Below is the cpu info of arm device (cat /proc/cpuinfo)

Below is lib list when ldd is used. (ignore libstdc++.so.6 -> not found as I am linking it using LD_LIBRARY_PATH and able to run it) 

**NOTE: same chip-tool binary is able to successfully commission and operate in same Linux device for WiFi matter plug (tapo P125M) with the 
[image]
[image]
same network.**

### Bug prevalence

Always. tried 6-7 times.

### GitHub hash of the SDK that was being used

1d62999ba2b6cc9f10bfe275a2b4e23da65c4f31,  and release SVE_23_03/rc2

### Platform

other

### Platform Version(s)

ARMv7 Processor rev 5 (v7l)

bzbarsky-apple, 2023-09-18:
> But after another iteration. i.e. reset the device, pair using the Google Home app, and commission using 2nd admin. via my custom Linux device using the same command which is on the same wifi network.

@EIC-deval-jariwala When the second commissioning window (the one after initial commissioning) is opened, does it get opened using OpenCommissioningWindow, or OpenBasicCommissioningWindow?  That is, is the `13787619912` code coming from the Home app or from the device packaging?

What does a DNS-SD browse for `_matterc._udp` show at the point when the commissioning is failing?  

Because at first glance, the most likely thing going on here is that the wrong set of IP addresses is being discovered, likely due to cached information from a previous advertisement...  But hard to tell for sure without either device logs or more information about the full set of things being advertised.

EIC-deval-jariwala, 2023-09-18:
@bzbarsky-apple Thanks for a quick replay.

Below are the answers to the questions you asked for:
1. The code 13787619912 is coming from the Google Home app after I click on 'settings->Linked Matter apps and services-> Link apps and services'. 

- As This code is random each time, I believe it should be using the Enhanced commissioning method internally. As for basic commissioning, it should have shared the same pairing code each time which the manufacturer provided with the device QR code.

2. Where do find ' DNS-SD browse for _matterc._udp' information. Did you mean to get it from the printing log when we execute chip tool in linux device? 

Below are logs:
- x86 machine as 2nd pairing device. (Success)
[thread_with_x86_2nd_admin_host_OK_with_google_nest_1.txt](https://github.com/project-chip/connectedhomeip/files/12653247/thread_with_x86_2nd_admin_host_OK_with_google_nest_1.txt)

- RPi-4B board as 2nd admin device. (Success)
[thread_with_rpi4_2nd_admin_host_OK_with_google_nest_1.txt](https://github.com/project-chip/connectedhomeip/files/12653251/thread_with_rpi4_2nd_admin_host_OK_with_google_nest_1.txt)

- Custom Linux board as 2nd admin device. (Failure)
[thread_with_customLinuxBoard_2nd_admin_NOT-OK_with_google_nest_1.txt](https://github.com/project-chip/connectedhomeip/files/12653256/thread_with_customLinuxBoard_2nd_admin_NOT-OK_with_google_nest_1.txt)

We can find from all the logs that the address for DNS-SD is common in all
..............
CHIP:CTL: Discovered device to be commissioned over DNS-SD
CHIP:CTL: Attempting PASE connection to UDP:[fd17:b440:895a:1:a507:7c1b:87c1:af21]:5540
.............
It is able to commission for x86 and RPi-4b but not for Custom-Linux-board, and rpi-zero2w board (log not attached).

EIC-deval-jariwala, 2023-09-19:
@bzbarsky-apple . Have you got chance to look at the logs ?

bzbarsky-apple, 2023-09-20:
> Have you got chance to look at the logs ?

I had not.  If I had, I would have commented here.

> I believe it should be using the Enhanced commissioning method internally.

Yes.

> Where do find ' DNS-SD browse for _matterc._udp' information

Using Avahi on Linux or the dns-sd utility on Mac.

Looking at those last success/failure logs, the "this device is now using a different IP" is still a possibility but looks less likely, since the other two logs both use the same IP.

So the real question then becomes: why do you never get a response to the PBKDFParamRequest message?  You didn't include the log for the commissionee, so it's hard to tell which of these is going on:

1. The PBKDFParamRequest never reaches the commissionee.
2. The commissionee does not respond to the PBKDFParamRequest.
3. The PBKDFParamResponse never reaches the commissioner.

I suggest looking at the commissionee logs to determine that.  And maybe checking whether the "custom Linux board" can ping the relevant comissionee IP.

EIC-deval-jariwala, 2023-09-21:
@bzbarsky-apple . Thanks again. Answers to your questions are as below.
1. The PBKDFParamRequest never reaches the commissionee. - that might be the case
As I tried pinging the IP-address and it didn't work for custom-board. It did worked for x86 and RPi-4B.

I tried the same sequence using RPi-Zero2w board and have same result as my Linux board.
Thought, same Os and chip-tool version are same for both. And RPi-Zero2w also didn't respond to ping.

I have a Question that is this IPv6 address is bound to end device via border router.? As during pinding if I unpug the end device it stops coming response in x86/RPi-4B and after plug it again It started responding.

bzbarsky-apple, 2023-09-21:
> As I tried pinging the IP-address and it didn't work for custom-board. It did worked for x86 and RPi-4B.

OK, that seems like the thing to pursue, then.

> is this IPv6 address is bound to end device via border router.

I don't know.