project-chip/connectedhomeip #29310: [Feature] Get the matter-stack version

caipiblack, 2023-09-18:
### Feature description

Hello,

Is it possible to get the stack matter version from the code ?

I see that there is "CHIPVersion.h" file generated during the build by a script called in chip_version.gni

So (maybe/not sure) currently we have to manually set these variables to get the correct version at build time in CHIPVersion.h:

```
  chip_version_major = 0
  chip_version_minor = 0
  chip_version_patch = 0
  chip_version_extra = ""
```

Are you planing to automatise it  ? For example based on git tag, or something else ?

### Platform

all

bzbarsky-apple, 2023-09-18:
There is no real concept of "stack version" at the moment.  There are constants for the interaction model and data model revision, but no other versioning going on other than the actual git SHA.

caipiblack, 2023-09-18:
What about the git SHA, is it possible to get it somewhere by calling a function ?

We want to get an information from our software to identify the version/revision/commit of the matter-stack

By version I mean "1.1.0.1" for exemple

bzbarsky-apple, 2023-09-18:
> By version I mean "1.1.0.1" for exemple

bzbarsky-apple, 2023-09-18:
> What about the git SHA, is it possible to get it somewhere by calling a function ?

At runtime?  I don't believe so.

> By version I mean "1.1.0.1" for exemple

I know what you mean, but those version numbers don't really exist for the SDK as things stand.

> And where are the constants for the interaction model and data model revision please ?

In `src/app/DataModelRevision.h` and `src/app/InteractionModelRevision.h`.

caipiblack, 2023-10-04:
I'm closing this issue as this doesn't seem to be usefull currently (because there is no real concept of version)

In our side, as we build statically with "libCHIP.a", we know the version/commit of the Matter stack (at build time) so we can store it a pre-processor definition and display it at runtime.