project-chip/connectedhomeip #29324: [Android]Figure out how to generate android chip-tool with setting matter_enable_java_generated_api as false

yunhanw-google, 2023-09-18:


joonhaengHeo, 2023-09-21:
@yunhanw-google 
Currently, this feature cannot be disabled due to the code in the folder below in Android ChipTool.
https://github.com/project-chip/connectedhomeip/tree/master/examples/android/CHIPTool/app/src/main/java/com/google/chip/chiptool/clusterclient/clusterinteraction

Also, there was a history of trying to change it through the PR below, but it was not merged due to the content of the comment.
https://github.com/project-chip/connectedhomeip/pull/28467

yunhanw-google, 2023-11-06:
decide not disabled since community is broadly using these generated API, we would replace the underneath implemenation with generic IM API

closed 2023-11-06 by yunhanw-google