project-chip/connectedhomeip #29330: Is chip-tool support pairing switchbot hub 2?

tkaewplik, 2023-09-19:
# Reproduction steps

I am sorry for my newbies here, I am trying to pair a Matter hub (RPI3 build from image from https://docs.silabs.com/matter/1.0.4/matter-overview/) with a Matter device (switchbot hub 2)

1. download and upload the RPI image to SD card (https://www.silabs.com/documents/public/software/SilabsMatterPi_2.0.0-1.1.zip)
2. Get the Matter setup-code from switchbot hub 2 (follow the tutorial here https://www.youtube.com/watch?v=OEQLU70tykE)
3. Run the pairing code below
```bash
cd /home/ubuntu/connectedhomeip/out/standalone
./chip-tool pairing code <node-id> <matter-setup-code-from-switchbot-hub-2>
```
After running the pairing code, I got the error and can not pair them, there are many errors in the log so I can not figure out which one is the main cause or just everything here is not supported yet?

I would appreciate it if you could guide me, do I understand something wrong about pairing devices and hub or is this a bug or unsupported feature that I have to wait for?

```log
ubuntu@ubuntu:~/connectedhomeip/out/standalone$ ./chip-tool pairing code 1 3524091xxxx
[1695114890.263351][5136:5136] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1695114890.280634][5136:5136] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1695114890.281261][5136:5136] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1695114890.281497][5136:5136] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1695114890.282257][5136:5136] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-Ln4IpN)
[1695114890.283607][5136:5136] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1695114890.283776][5136:5136] CHIP:DL: NVS set: chip-counters/reboot-count = 7 (0x7)
[1695114890.284962][5136:5136] CHIP:DL: Got Ethernet interface: eth0
[1695114890.285799][5136:5136] CHIP:DL: Found the primary Ethernet interface:eth0
[1695114890.286715][5136:5136] CHIP:DL: Got WiFi interface: wlan0
[1695114890.286896][5136:5136] CHIP:DL: Failed to reset WiFi statistic counts
[1695114890.287082][5136:5136] CHIP:IN: UDP::Init bind&listen port=0
[1695114890.287310][5136:5136] CHIP:IN: UDP::Init bound to port=34311
[1695114890.287355][5136:5136] CHIP:IN: UDP::Init bind&listen port=0
[1695114890.287536][5136:5136] CHIP:IN: UDP::Init bound to port=49055
[1695114890.287579][5136:5136] CHIP:IN: BLEBase::Init - setting/overriding transport
[1695114890.287618][5136:5136] CHIP:IN: TransportMgr initialized
[1695114890.287680][5136:5136] CHIP:FP: Initializing FabricTable from persistent storage
[1695114890.287942][5136:5136] CHIP:TS: Last Known Good Time: 2022-11-01T16:49:03
[1695114890.291245][5136:5136] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x31D95E9E2C7EDF3B, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1695114890.295433][5136:5136] CHIP:ZCL: Using ZAP configuration...
[1695114890.299083][5136:5136] CHIP:DL: MDNS failed to join multicast group on wlan0 for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:761: Inet Error 0x00000110: Address not found
[1695114890.301282][5136:5136] CHIP:CTL: System State Initialized...
[1695114890.301510][5136:5136] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1695114890.301630][5136:5136] CHIP:CTL: Setting attestation nonce to random value
[1695114890.301729][5136:5136] CHIP:CTL: Setting CSR nonce to random value
[1695114890.301843][5136:5136] CHIP:IN: UDP::Init bind&listen port=5550
[1695114890.302049][5136:5136] CHIP:IN: UDP::Init bound to port=5550
[1695114890.302093][5136:5136] CHIP:IN: UDP::Init bind&listen port=5550
[1695114890.302247][5136:5136] CHIP:IN: UDP::Init bound to port=5550
[1695114890.302296][5136:5136] CHIP:IN: TransportMgr initialized
[1695114890.302945][5136:5141] CHIP:DL: CHIP task running
[1695114890.303950][5136:5141] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1695114890.304049][5136:5141] CHIP:CTL: Setting attestation nonce to random value
[1695114890.304552][5136:5141] CHIP:CTL: Setting CSR nonce to random value
[1695114890.306673][5136:5141] CHIP:CTL: Generating NOC
[1695114890.308711][5136:5141] CHIP:FP: Validating NOC chain
[1695114890.312342][5136:5141] CHIP:FP: NOC chain validation successful
[1695114890.312923][5136:5141] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1695114890.312994][5136:5141] CHIP:TS: Last Known Good Time: 2022-11-01T16:49:03
[1695114890.313038][5136:5141] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1695114890.313080][5136:5141] CHIP:TS: Retaining current Last Known Good Time
[1695114890.317159][5136:5141] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1695114890.321467][5136:5141] CHIP:TS: Committing Last Known Good Time to storage: 2022-11-01T16:49:03
[1695114890.325643][5136:5141] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1695114890.325765][5136:5141] CHIP:IN: UDP::Init bind&listen port=5550
[1695114890.326007][5136:5141] CHIP:IN: UDP::Init bound to port=5550
[1695114890.326054][5136:5141] CHIP:IN: UDP::Init bind&listen port=5550
[1695114890.326213][5136:5141] CHIP:IN: UDP::Init bound to port=5550
[1695114890.326254][5136:5141] CHIP:IN: TransportMgr initialized
[1695114890.348315][5136:5141] CHIP:CTL: Setting attestation nonce to random value
[1695114890.348466][5136:5141] CHIP:CTL: Setting CSR nonce to random value
[1695114890.348597][5136:5141] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[... 22 lines cut ...]
[1695114920.355579][5136:5136] CHIP:CTL: Shutting down the commissioner
[1695114920.355661][5136:5136] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1695114920.356042][5136:5136] CHIP:CTL: Shutting down the controller
[1695114920.356124][5136:5136] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1695114920.357990][5136:5136] CHIP:DMG: IM WH moving to [Uninitialized]
[1695114920.358090][5136:5136] CHIP:DMG: IM WH moving to [Uninitialized]
[1695114920.358168][5136:5136] CHIP:DMG: IM WH moving to [Uninitialized]
[1695114920.358263][5136:5136] CHIP:DMG: IM WH moving to [Uninitialized]
[1695114920.358344][5136:5136] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1695114920.358666][5136:5136] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1695114920.359563][5136:5136] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-VIp6bu)
[1695114920.362097][5136:5136] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1695114920.362343][5136:5136] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1695114920.362448][5136:5136] CHIP:DL: Inet Layer shutdown
[1695114920.362532][5136:5136] CHIP:DL: BLE shutdown
[1695114920.362611][5136:5136] CHIP:DL: System Layer shutdown
[1695114920.363308][5136:5136] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/pairing/PairingCommand.cpp:164: CHIP Error 0x00000003: Incorrect state
```

# Platform
- Raspberry PI 3 Model B V1.2
- Switchbot hub 2

# Version
- commit: 7f2b0a5920d1a7acc202c2932ac4c626c54cd31e

bzbarsky-apple, 2023-09-19:
@tkaewplik The main error here is:
```
[1695114920.352856][5136:5141] CHIP:CTL: Discovery timed out
```
which means chip-tool never saw the commissionable advertisement for the device being paired.

This could happen for various reasons.  Things to check:

1. Is the device being paired already on the IP network?  If it is, it should be advertising via DNS-SD; if not, via BLE.
2. If the device being paired is _not_ on the IP network, then `pairing code` is the wrong command to use for chip-tool, since that only does on-network pairing.
3. If the device being paired _is_ on the IP network, is it advertising there (as in, is the commissioning window on it actually open?).

Start with those.

tkaewplik, 2023-09-24:
Thank you for your explanation

I paired my device to IP network already but not by chip-tool pairing ble-wifi command, I have used the switchbot hub 2 application on mobile for it and can see it on my router page and also can ping to it and maybe exist on `chip-tool discover commissionables` command (please see the below detail). Could this be the reason? so, do I have to pair it using chip-tool BLE only?

1.,2. Currently, I don't know the setup pin code for BLE pairing of my device so I am asking the switchbot hub 2 support center about it, and then I will try pairing it again.
3. Did you mean this command? (if yes, It's also an error)
  ```
  $ ./chip-tool pairing open-commissioning-window <node_id> <option> <window_timeout> <iteration> <discriminator>
  ```
in the switchbot hub2 instructions for Matter I have to hold the on and off button for 15 seconds, maybe this is about the commissioning window? but after doing this, the device will be reset, and can not remember WIFI, and I have to configure the WIFI via mobile again and still it can not be paired using `pairing code`.

I am sorry about my headache issue, I would appreciate it if you could give me advice. 

# more information
- I tried running the discover commissionables command as you see below, It seems that It can detect the device but still has some error on the log.

```
ubuntu@ubuntu:~/connectedhomeip/out/standalone$ ./chip-tool discover commissionables
[1695533494.045999][3982:3982] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1695533494.059862][3982:3982] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1695533494.060445][3982:3982] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1695533494.060727][3982:3982] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1695533494.061717][3982:3982] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-CIN11c)
[1695533494.063117][3982:3982] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1695533494.063421][3982:3982] CHIP:DL: NVS set: chip-counters/reboot-count = 8 (0x8)
[1695533494.065123][3982:3982] CHIP:DL: Got Ethernet interface: eth0
[1695533494.065989][3982:3982] CHIP:DL: Found the primary Ethernet interface:eth0
[1695533494.066920][3982:3982] CHIP:DL: Got WiFi interface: wlan0
[1695533494.067056][3982:3982] CHIP:DL: Failed to reset WiFi statistic counts
[1695533494.067224][3982:3982] CHIP:IN: UDP::Init bind&listen port=0
[1695533494.067489][3982:3982] CHIP:IN: UDP::Init bound to port=49830
[1695533494.067539][3982:3982] CHIP:IN: UDP::Init bind&listen port=0
[1695533494.067718][3982:3982] CHIP:IN: UDP::Init bound to port=55799
[1695533494.067761][3982:3982] CHIP:IN: BLEBase::Init - setting/overriding transport
[1695533494.067808][3982:3982] CHIP:IN: TransportMgr initialized
[1695533494.067869][3982:3982] CHIP:FP: Initializing FabricTable from persistent storage
[1695533494.068112][3982:3982] CHIP:TS: Last Known Good Time: 2022-11-01T16:49:03
[1695533494.071411][3982:3982] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x7827930E14F86B83, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1695533494.075543][3982:3982] CHIP:ZCL: Using ZAP configuration...
[1695533494.079069][3982:3982] CHIP:DL: MDNS failed to join multicast group on wlan0 for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:761: Inet Error 0x00000110: Address not found
[1695533494.081566][3982:3982] CHIP:CTL: System State Initialized...
[1695533494.081709][3982:3982] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1695533494.081842][3982:3982] CHIP:CTL: Setting attestation nonce to random value
[1695533494.081954][3982:3982] CHIP:CTL: Setting CSR nonce to random value
[1695533494.082071][3982:3982] CHIP:IN: UDP::Init bind&listen port=5550
[1695533494.082350][3982:3982] CHIP:IN: UDP::Init bound to port=5550
[1695533494.082405][3982:3982] CHIP:IN: UDP::Init bind&listen port=5550
[1695533494.082632][3982:3982] CHIP:IN: UDP::Init bound to port=5550
[1695533494.082678][3982:3982] CHIP:IN: TransportMgr initialized
[1695533494.083640][3982:3987] CHIP:DL: CHIP task running
[1695533494.084633][3982:3987] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1695533494.084742][3982:3987] CHIP:CTL: Setting attestation nonce to random value
[1695533494.085139][3982:3987] CHIP:CTL: Setting CSR nonce to random value
[1695533494.087748][3982:3987] CHIP:CTL: Generating NOC
[1695533494.089833][3982:3987] CHIP:FP: Validating NOC chain
[1695533494.093587][3982:3987] CHIP:FP: NOC chain validation successful
[1695533494.093998][3982:3987] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1695533494.094073][3982:3987] CHIP:TS: Last Known Good Time: 2022-11-01T16:49:03
[1695533494.094117][3982:3987] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1695533494.094159][3982:3987] CHIP:TS: Retaining current Last Known Good Time
[1695533494.098512][3982:3987] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1695533494.102531][3982:3987] CHIP:TS: Committing Last Known Good Time to storage: 2022-11-01T16:49:03
[1695533494.106344][3982:3987] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1695533494.106530][3982:3987] CHIP:IN: UDP::Init bind&listen port=5550
[1695533494.106783][3982:3987] CHIP:IN: UDP::Init bound to port=5550
[... cut ...]
[1695533524.092811][3982:3982] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1695533524.093442][3982:3982] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1695533524.094431][3982:3982] CHIP:DL: Inet Layer shutdown
[1695533524.094524][3982:3982] CHIP:DL: BLE shutdown
[1695533524.094618][3982:3982] CHIP:DL: System Layer shutdown
[1695533524.094941][3982:3982] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:513: CHIP Error 0x00000032: Timeout
```

tkaewplik, 2023-09-26:
# Update
I have tried again with `pairing code-wifi` instead so the `Incorrect state` error is gone.
```
./chip-tool pairing code-wifi 1 <wifi_name> <wifi_pass> 3524091xxxx
```

However, I got a new error (internal error) about `AttestationVerification` step.

```
[1695727200.486979][1596:1596] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs

[1695727200.500510][1596:1596] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini

[1695727200.501035][1596:1596] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini

[1695727200.501306][1596:1596] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini

[1695727200.502070][1596:1596] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-Haob8F)

[1695727200.503408][1596:1596] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)

[1695727200.503542][1596:1596] CHIP:DL: NVS set: chip-counters/reboot-count = 5 (0x5)

[1695727200.504842][1596:1596] CHIP:DL: Got Ethernet interface: eth0

[1695727200.505685][1596:1596] CHIP:DL: Found the primary Ethernet interface:eth0

[1695727200.506661][1596:1596] CHIP:DL: Got WiFi interface: wlan0

[1695727200.506818][1596:1596] CHIP:DL: Failed to reset WiFi statistic counts

[1695727200.507001][1596:1596] CHIP:IN: UDP::Init bind&listen port=0

[1695727200.507235][1596:1596] CHIP:IN: UDP::Init bound to port=59070

[1695727200.507287][1596:1596] CHIP:IN: UDP::Init bind&listen port=0

[1695727200.507474][1596:1596] CHIP:IN: UDP::Init bound to port=47100

[1695727200.507519][1596:1596] CHIP:IN: BLEBase::Init - setting/overriding transport

[1695727200.507584][1596:1596] CHIP:IN: TransportMgr initialized

[1695727200.507651][1596:1596] CHIP:FP: Initializing FabricTable from persistent storage

[1695727200.507907][1596:1596] CHIP:TS: Last Known Good Time: 2022-11-01T16:49:03

[1695727200.511180][1596:1596] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x88F133B8A929FAA1, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1

[1695727200.515638][1596:1596] CHIP:ZCL: Using ZAP configuration...

[1695727200.519551][1596:1596] CHIP:DL: MDNS failed to join multicast group on wlan0 for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:761: Inet Error 0x00000110: Address not found

[1695727200.521938][1596:1596] CHIP:CTL: System State Initialized...

[1695727200.522063][1596:1596] CHIP:CTL: Stopping commissioning discovery over DNS-SD

[1695727200.522245][1596:1596] CHIP:CTL: Setting attestation nonce to random value

[1695727200.522350][1596:1596] CHIP:CTL: Setting CSR nonce to random value

[1695727200.522502][1596:1596] CHIP:IN: UDP::Init bind&listen port=5550

[1695727200.522750][1596:1596] CHIP:IN: UDP::Init bound to port=5550

[1695727200.522799][1596:1596] CHIP:IN: UDP::Init bind&listen port=5550

[1695727200.522989][1596:1596] CHIP:IN: UDP::Init bound to port=5550

[1695727200.523033][1596:1596] CHIP:IN: TransportMgr initialized
[... 1212 lines cut ...]

[1695727207.290080][1596:1596] CHIP:DMG: IM WH moving to [Uninitialized]

[1695727207.290213][1596:1596] CHIP:DMG: IM WH moving to [Uninitialized]

[1695727207.290268][1596:1596] CHIP:DMG: IM WH moving to [Uninitialized]

[1695727207.290321][1596:1596] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet

[1695727207.290568][1596:1596] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.

[1695727207.291695][1596:1596] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-cMCm2H)

[1695727207.293819][1596:1596] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)

[1695727207.294610][1596:1596] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)

[1695727207.294758][1596:1596] CHIP:DL: Inet Layer shutdown

[1695727207.294817][1596:1596] CHIP:DL: BLE shutdown

[1695727207.297453][1596:1596] CHIP:DL: System Layer shutdown

[1695727207.297730][1596:1596] CHIP:TOO: Run command failure: ../../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1031: CHIP Error 0x000000AC: Internal error
```

tkaewplik, 2023-09-27:
I found out that AttestationVerificationResult err 101 is about kPaaNotFound so is it possible that it is the old version issue?

tkaewplik, 2023-09-27:
Execute me, Is re-compiling the project again the only way to update the PAA of chip-tool?

bzbarsky-apple, 2023-09-28:
@tkaewplik First, please _attach_ logs, not paste them.  This issue is now incredibly hard to work with.

Second, that commissioning log sure looks like you are connecting to the device over BLE, not over the IP network.  It does look like the numeric code you are using is a possible match for the discriminator "discover commissionables" showed, so maybe this device is advertising both over BLE and over DNS-SD even though it's on the IP network (a spec violation: it should not be advertising over BLE at this point).

Third: kPaaNotFound makes sense if you did not pass any actual non-test PAAs to chip-tool.  By default it only recognizes the test PAAs; see the `--paa-trust-store-path` argument for telling it to do something else.

tkaewplik, 2023-09-28:
@bzbarsky-apple My apologies for not attaching the log as files, I have revised them all to log files already.

Thank you for your guideline, I think, `--paa-trust-store-path` allows the device to connect to chip-tool now.
```bash
$ ./chip-tool pairing code-wifi 1 <wifi-name> <wifi-pass> 352409183xx --paa-trust-store-path credentials/development/paa-root-certs
```
[paired.log](https://github.com/project-chip/connectedhomeip/files/12746166/paired.log)

However, I got messages`no endpoint` after pairing and could not find a supported cluster from any endpoints. (got UNSUPPORTED_CLUSTER when tried to send onoff and identify read to random endpoint id)

Could I ask a simple question here? how can I know the endpoint of the node? Does this mean my device has no endpoint?

The message after pairing, shows no endpoint error
```
CHIP:CTL: Successfully finished commissioning step 'Cleanup'
CHIP:TOO: Device commissioning completed with success
CHIP:DMG: ICR moving to [AwaitingDe]
CHIP:EM: <<< [E:18201i S:26823 M:217925161 (Ack:231275443)] (S) Msg TX to 1:0000000000000001 [1CD3] [UDP:[fe80::424c:caff:fe7c:9ab8%enp0s31f6]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)
CHIP:EM: Flushed pending ack for MessageCounter:231275443 on exchange 18201i
CHIP:DL: HandlePlatformSpecificBLEEvent 16388
CHIP:BLE: no endpoint for unsub complete
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:BLE: no endpoint for BLE sent data ack
CHIP:CTL: Shutting down the commissioner
CHIP:CTL: Shutting down the controller
```

bzbarsky-apple, 2023-09-28:
> The message after pairing, shows no endpoint error

This is about BLE endpoints.  The BLE connection got shut down, so this is expected; it's not an error condition....

> and could not find a supported cluster from any endpoints. (got UNSUPPORTED_CLUSTER when tried to send onoff and identify read to random endpoint id)

Well, why are you using a random endpoint id?  What endpoints (this is not Matter endpoints, which is unrelated to BLE endpoints) does the device expose?  It should always have endpoint 0.  You can reads the PartsList attribute from the Descriptor cluster on endpoint 0 to get a list of the other endpoints the device has, and then you can read various attributes from Descriptor clusters on those other endpoints to find out what device types they claim, what clusters are supported on those endpoints, etc.

closed 2023-09-28 by tkaewplik