project-chip/connectedhomeip #29331: [BUG] TC-DGGEN-2.1 issues

jjm202, 2023-09-19:
### Reproduction steps

Execute TC-DGGEN-2.1. 

With the enableKey enabled, note that a test-event-trigger value of 0x3 is sent to the DUT. What if the DUT doesn't support 0x3 but a set of other trigger values? In other words, what derived a trigger-value of 0x000000000000003? Does it need to be handled in DUT firmware just to pass this TC step? 

`CHIP:TOO Command: generaldiagnostics test-event-trigger hex:00112233445566778899aabbccddeeff 0x0000000000000003 0x394ef324f5c9788c 0`

Also, the user is not directed to recommission the DUT after a factory reset in Step 10b, which causes retrieving the DUT to fail:

`Wait for the commissioned device to be retrieved` 

The proper dialogs should be added to inform the tester the device needs to be recommissioned. 

### Bug prevalence

always

### GitHub hash of the SDK that was being used

latest

### Platform

raspi

### Platform Version(s)

test harness

bzbarsky-apple, 2023-09-19:
> what derived a trigger-value of 0x000000000000003?

That's the default value that is used by the example apps, but this should be configurable to the actual value the device uses (see TEST_EVENT_TRIGGER_KEY), just like the enable key is configurable via TEST_EVENT_ENABLE_KEY.

> Also, the user is not directed to recommission the DUT after a factory reset in Step 10b

This is an issue for step 1a, too, looks like.  The test plan has this right; it's the test script that's broken.  @manjunath-grl please take a look.

manjunath-grl, 2023-09-21:
@jjm202 We will update the prompt logs with the comment mentioning to  
```
 Reset DUT to factory defaults
 Set the DUT to Discoverable mode
```
@bzbarsky-apple  What is your thoughts on this?

bzbarsky-apple, 2023-09-21:
@manjunath-grl My thoughts are that those are not steps that match the test plan.  The issue is not that you have to "set to discoverable mode".  You have to _commission the device_.