project-chip/connectedhomeip #29336: Matter Controller on Linux

LongND0105, 2023-09-20:
I'am developing a Matter controller for raspberry pi. This device receives message from Cloud MQTT and then run them for commissioning, on/off, v.v. Could I use `connectedhomeip/example/chip-tool` example for dev?

bzbarsky-apple, 2023-09-20:
@LongND0105 chip-tool is designed as a testing tool, in general.  But if it serves your needs, you could obviously use it however you like.

What is the actual question here?

LongND0105, 2023-09-21:
I don't understand the code architecture of the chip-tool, so I don't know how to make it run as an event loop to receive control commands from the cloud server and then execute functions with those commands. I also don't understand what local storage is for or why we need to initialize memory. Can you explain this to me? I want to have an overview of how Matter projects work and the specific requirements to ensure it doesn't encounter segmentation faults (core dumps)."

This statement expresses a desire to understand the architecture and functionality of a software tool (chip-tool) used in a technology project. The person is seeking explanations for concepts like event loops, handling commands from a cloud server, local storage usage, and memory initialization. They are also interested in gaining a comprehensive understanding of how Matter projects operate and how to prevent common issues like segmentation faults.

firebladed, 2023-09-24:
Im not sure from the original exactly what is wanted,

However from my perspective, unless ive missed something, there is little information on how you would go from using the chip tool, fine for testing, to an actual controller implementation on Linux.

e.g if I have raspberry pi running open border router  and a nordic thingy 53 running matter weather station
i can use the chip tool to commission and query them.

but how do I go from that to say a simple C/C++ app that talks to the thingy53 and prints out the temperature, humidity, pressure and battery level.

I'm currently thinking I'd prefer if the chip-tool was in two parts, a library with all the command implementations and a cli front end with all the user interaction.

you could then could base various controller implementations off the chip-tool library.

it currently looks like you would have to re-implement various chunks of the chip-tool to construct a basic controller on linux

bzbarsky-apple, 2023-10-03:
> so I don't know how to make it run as an event loop to receive control commands from the cloud server and then execute functions with those commands

You could run `chip-tool interactive server` and then send JSON messages to chip-tool on the socket it listens on...  But this may be more work than just setting up your own client using the SDK directly, for the use cases you care about.

> I also don't understand what local storage is for or why we need to initialize memory. 

I don't know what you mean by "local storage" and what sort of memory initialization you're talking about, so hard to say.

> I want to have an overview of how Matter projects work

There isn't one already, unfortunately, apart from reading the various headers and the existing documentation under `docs`.

> and the specific requirements to ensure it doesn't encounter segmentation faults

Those are ... wide-ranging, depending on what you are doing.

bzbarsky-apple, 2023-10-03:
> a library with all the command implementations

The "command implementations" in chip-tool (if you ignore all the complexity involved in dealing with invalid things, etc, which chip-tool has to support as a testing tool) are just glue going from the cli front-end to fairly simple APIs exposed by src/controller, no?