project-chip/connectedhomeip #29337: source scripts/bootstrap.sh, Setting up Python environment.....[\]Error:

qq8512852, 2023-09-20:
### Build issue(s)

## 1. Basic Information
Git Branch: remotes/origin/test_event_7
Hardware: Raspberry PI 4B  (Linux wubo-desktop 5.15.0-1037-raspi #40-Ubuntu SMP PREEMPT Fri Aug 25 16:42:00 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux)  
OS: ubuntu 22.04 desktop 64bit
Python version:  Python 3.8.18 (default, Aug 25 2023, 13:20:30)
 

## 2. Log Print
WELCOME TO...

  ▄███▒  ░▓█  ░▓█ ░▓█▓ ▒█████▄
 ██▒ ▀█▒  ▒█   ▒█  ░█▒  ▒█░  █░
 █▓░      ▒██████  ░█▒  ▒█▄▄▄█░
 ▓█   █▒  ▒█   ▒█  ░█░  ▒█▀
 ░▓███▀  ░▓███░▓█▒ ░█░  ▒█

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (1m22.2s)
  Setting up Python environment.....[\]Error: Command '['/home/wubo/Desktop/matter/git-hub-matter/connectedhomeip/.environment/pigweed-venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']' returned non-zero exit status 1.
['/usr/bin/python3', '-m', 'venv', '--upgrade', '/home/wubo/Desktop/matter/git-hub-matter/connectedhomeip/.environment/pigweed-venv'] {'env': {'SHELL': '/bin/bash', 'PW_DOCTOR_SKIP_CIPD_CHECKS': '1', 'LC_ADDRESS': 'zh_CN.UTF-8', 'LC_NAME': 'zh_CN.UTF-8', 'PW_PROJECT_ROOT': '/home/wubo/Desktop/matter/git-hub-matter/connectedhomeip', 'LC_MONETARY': 'zh_CN.UTF-8', 'PWD': '/home/wubo/Desktop/matter/git-hub-matter/connectedhomeip', 'LOGNAME': 'wubo', 'XDG_SESSION_TYPE': 'tty', 'MOTD_SHOWN': 'pam', 'PW_ROOT': '/home/wubo/Desktop/matter/git-hub-matter/connectedhomeip/third_party/pigweed/repo', 'HOME': '/home/wubo', 'LC_PAPER': 'zh_CN.UTF-8', 'LANG': 'en_US.UTF-8', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:', 'SSH_CONNECTION': '192.168.50.54 24640 192.168.50.225 22', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'XDG_SESSION_CLASS': 'user', 'LC_IDENTIFICATION': 'zh_CN.UTF-8', 'TERM': 'xterm', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'USER': 'wubo', 'DISPLAY': 'localhost:10.0', 'SHLVL': '1', 'LC_TELEPHONE': 'zh_CN.UTF-8', 'LC_MEASUREMENT': 'zh_CN.UTF-8', 'XDG_SESSION_ID': '38', 'XDG_RUNTIME_DIR': '/run/user/1000', '_PW_ACTUAL_ENVIRONMENT_ROOT': '/home/wubo/Desktop/matter/git-hub-matter/connectedhomeip/.environment', 'PW_BRANDING_BANNER': '/home/wubo/Desktop/matter/git-hub-matter/connectedhomeip/scripts/setup/banner.txt', 'SSH_CLIENT': '192.168.50.54 24640 22', 'LC_TIME': 'zh_CN.UTF-8', 'XDG_DATA_DIRS': '/usr/share/gnome:/usr/local/share:/usr/share:/var/lib/snapd/desktop', 'PATH': '/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus', 'SSH_TTY': '/dev/pts/0', 'LC_NUMERIC': 'zh_CN.UTF-8', 'OLDPWD': '/home/wubo/Desktop/matter/git-hub-matter/connectedhomeip/third_party', '_': '/usr/bin/python3', 'CIPD_HTTP_USER_AGENT_PREFIX': 'pigweed-infra/tools/f0e9205b4185fdd1409444e5636a8f955a441971'}, 'stdout': <_io.TextIOWrapper name=3 mode='w+' encoding='UTF-8'>, 'stderr': -2}
None
Traceback (most recent call last):
  File "/home/wubo/Desktop/matter/git-hub-matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 679, in <module>
    sys.exit(main())
  File "/home/wubo/Desktop/matter/git-hub-matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 671, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/wubo/Desktop/matter/git-hub-matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 414, in setup
    result = step(spin)
  File "/home/wubo/Desktop/matter/git-hub-matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 533, in virtualenv
    if not virtualenv_setup.install(
  File "/home/wubo/Desktop/matter/git-hub-matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 195, in install
    _check_call(cmd, env=envcopy)
  File "/home/wubo/Desktop/matter/git-hub-matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 87, in _check_call
    subprocess.check_call(args, **kwargs)
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'venv', '--upgrade', '/home/wubo/Desktop/matter/git-hub-matter/connectedhomeip/.environment/pigweed-venv']' returned non-zero exit status 1.
wubo@wubo-desktop:~/Desktop/matter/git-hub-matter/connectedhomeip$

How to fix it?

### Platform

raspi

andy31415, 2023-09-20:
Seems like libev error, we had this because gdbgui dependency in esp32.

Please do the following changes:
  - remove (comment out with) gdbgui from `scripts/setup/requirements.esp32.txt`
  - if esp32 compile needed, you can comment out gdbgui from `esp-idf/requirements.txt` from wherever you have espidf installed

andy31415, 2023-09-20:
Having said that, above you say `Git Branch: remotes/origin/test_event_7` ... this is a 2 year old branch. Why are you using it?

https://github.com/project-chip/connectedhomeip/commits/test_event_7

qq8512852, 2023-09-20:
> Having said that, above you say `Git Branch: remotes/origin/test_event_7` ... this is a 2 year old branch. Why are you using it?
> 
> https://github.com/project-chip/connectedhomeip/commits/test_event_7

Oh, because I first started working with Matter source code over a year ago, I have been using that version's codebase. After a long break, I‘m revisting it now and trying to compile it on a Raspberry Pi , but encountering difficulties. I 'm hesitant to switch branches as it may introduce further complications. But once environment is set up and  ready. I'll check the latest version.

qq8512852, 2023-09-20:
> Seems like libev error, we had this because gdbgui dependency in esp32.
>
> Please do the following changes:
>
> * remove (comment out with) gdbgui from `scripts/setup/requirements.esp32.txt`
> * if esp32 [...]

Thanks, I will try this solution on the device tomorrow morning.

closed 2023-09-21 by qq8512852