project-chip/connectedhomeip #29338: [BUG] Trouble with OTA Update on  Apple Home

yurenlee, 2023-09-20:
### Reproduction steps

Always

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

v1.1.0.1

### Platform

esp32

### Anything else?

The Matter device can't receive the OTA update from Apple Home. I've already performed the following checks reference to `Apple-Matter-OTA-User-Guide.pdf`.
* The controllers are an iPad Pro with iPadOS 16.6 and an iPhone 14 Pro with iOS 16.6. Both of these devices have the TestNet DCL Profile installed.
* The primary hub is an Apple TV 4K 2nd Gen, which also has the TestNet DCL Profile installed.
* It has been tested with a Linux OTA provider and can be updated normally.
* I've rebooted all devices and waited for at least 48 hours.

Other information

```
Device Model Version
Vendor ID: 4718 (0x126e)
Product ID: 1 (0x1)
Software Version: 2
Software Version String: v2.0
CD Version Number: 1
Firmware Information:

Software Version Valid: true
OTA URL: https://cdn.cnbj1.fds.api.mi-img.com/embedded-release/matter_ot_lib/xiaomi_dual_connectivity_v2.0.bin
OTA File Size: 1548303
OTA Checksum: IoED/Yz9d6HAB1DloUaMruQU7Pjl2n5jazzVAZvX2zk=
OTA Checksum Type: 1
Min Applicable Software Version: 1
Max Applicable Software Version: 1
Release Notes URL:

yurenlee, 2023-09-20:
As Apple developer requirements, feedback on the Feedback Assistant has been completed. Additionally, logs for the Apple TV homd process also has been attached.
https://feedbackassistant.apple.com/feedback/13190379

yurenlee, 2023-09-20:
Thanks for the quick reply,  I've reuploaded the attachments.

bzbarsky-apple, 2023-09-20:
@yurenlee Thank you.  So the OTA requestor log I see there does not have any QueryImage commands in it, unless I am missing something.  It also has node IDs for the requestor that I never see show up anywhere in the AppleTV log.

I do see the AppleTV getting several QueryImage commands, all from a node ID that does not appear in the OTA requestor log.  It responds to them, but it's not logging what it responds with, and there is never a BDX attempt....

Knowing what the responses to QueryImage looked like from the OTA requestor side would be pretty helpful....

yurenlee, 2023-09-21:
@bzbarsky-apple 
I apologize for the confusion. The logs I previously provided contained entries from various timestamps. I have since recaptured the logs, starting from the commissioning process and manually triggering an OTA query after all devices have rebooted. Both of those can be found at the same link.

I find this information quite valuable. 

From the OTA requestor's perspective, it received a 'QueryImageResponse,' indicating that the OTA provider was busy. 

* `esp32_ota_requestor_commissioning_ota_query.log`#Line.23339

```bash
I (281696) chip[EM]: <<< [E:58731i S:36765 M:50915488 (Ack:69743744)] (S) Msg TX to 1:6C9F59E95613BC32 [8037] --- Type 0000:10 (SecureChannel:StandaloneAck)
--
D (299656) chip[DMG]: 				{
D (299656) chip[DMG]: 					EndpointId = 0x0,
D (299656) chip[DMG]: 					ClusterId = 0x29,
D (299666) chip[DMG]: 					CommandId = 0x1,
D (299676) chip[DMG]: 				},
D (299676) chip[DMG]: 			
D (299676) chip[DMG]: 				CommandFields = 
D (299686) chip[DMG]: 				{
D (299686) chip[DMG]: 					0x0 = 1, 
D (299686) chip[DMG]: 					0x1 = 120, 
D (299686) chip[DMG]: 				},
D (299696) chip[DMG]: 			},
D (299696) chip[DMG]: 		
D (299696) chip[DMG]: 		},
D (299706) chip[DMG]: 	
D (299706) chip[DMG]: 	],
D (299706) chip[DMG]: 
D (299716) chip[DMG]: 	InteractionModelRevision = 1
D (299716) chip[DMG]: },
I (299716) chip[DMG]: Received Command Response Data, Endpoint=0 Cluster=0x0000_0029 Command=0x0000_0001
D (299726) chip[SWU]: QueryImageResponse:
D (299726) chip[SWU]:  status: 1
D (299726) chip[SWU]:  delayedActionTime: 120 seconds
I (299746) chip[SWU]: Scheduling a retry; delay: 120

```

Please let me know if any additional information is required.

bzbarsky-apple, 2023-09-21:
@yurenlee Thanks!  It does look like the AppleTV keeps responding Busy.  I am not sure why yet, but I asked some people to look into it....

bzbarsky-apple, 2023-09-22:
@yurenlee It looks like the issue may be here:

> Software Version String: v2.0

Currently Apple Home only supports version string of the form "x.y.z" (though that is a known issue that's being worked on).  Does it help if you change the version string to "2.0.0"?

yurenlee, 2023-09-22:
@bzbarsky-apple Thank you. I'll retest as soon as possible.

yurenlee, 2023-09-22:
@bzbarsky-apple I just completed a test, and the status has changed. It was previously 1, indicating 'Busy,' but it's now 2, indicating 'Not Available.' I'm not sure if I need to wait for some more time. I'll provide feedback in more than 12 hours.

yurenlee, 2023-09-25:
Hi, @bzbarsky-apple.
I have been waiting for over 48 hours, but I still can't receive the OTA update. The status code is 0x02. I removed the Matter accessory from the Home App and captured the logs again for your reference. For comparison, I also commissioned the same device to the Matter test fabric with a Linux OTA provider, and it was successful.

I have uploaded the logs to the same link. https://feedbackassistant.apple.com/feedback/13190379 .

Please let me know if you need any additional information.

bzbarsky-apple, 2023-09-28:
@yurenlee Sorry for the lag.  People who know what's going on with the Home side are looking into this, last I checked....

bzbarsky-apple, 2023-09-28:
@yurenlee I am being told that the issue has been sorted out.  Does that match your understanding of where we are?

yurenlee, 2023-09-28:
@bzbarsky-apple Yes, it is. Your Chinese colleague helped me resolve this issue. As he explained, the issue was caused by me re-editing the model version, which led to a cache exception on your server. He cleared all the cache, and now everything is working fine. Thank you very much. Best wishes.

silabs-DungAnh, 2025-05-15:
Hi @bzbarsky-apple, I saw this issue was marked as resolved. I'm still encountering a very similar issue that I can't update OTA with my Homepod Mini after I made a small change on my model version on TestNet DCL and I wanted to check if this is expected or if I should open a new issue?

dungdz16, 2025-05-15:
Hi @bzbarsky-apple, I saw this issue was marked as resolved. I'm still encountering a very similar issue that I can't update OTA with my Homepod Mini after I made a small change on my model version on TestNet DCL and I wanted to check if this is expected or if I should open a new issue?

bzbarsky-apple, 2025-05-20:
@dungdz16 I don't know; you should open a new issue, or more precisely a new feedback ticket following the instructions at https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#providing-feedback-to-apple

lunDreame, 2025-08-10:
@yurenlee 
Hello. I have a few questions. Is it at least 24 hours for Apple Home to get an alarm? Or is it possible to get an alarm earlier? And if the device goes offline or reboots in the middle, does the time reset again? Thank you.