project-chip/connectedhomeip #29359: [BUG] DataReports with EnableTagCompression can not be parsed

Apollon77, 2023-09-20:
### Reproduction steps

According to 1.0 and 1.1 Specs the AttributeReportIB used in DataReports can use Tag compression. But when I do this then chip tool reports

> Error: src/app/ReadClient.cpp:643: CHIP Error 0x000000B5: Malformed Interacton Model Attribute Path IB

When I follow the code this expects a "GetConcreteAttributePath" and in this method hard reads and endpoint-id from the data which is not there in the compressed DataReport.
https://github.com/project-chip/connectedhomeip/blob/master/src/app/ReadClient.cpp#L645

TagCompression is _not_ listed as provisional specification feature in Matter 1.0 or 1.1 (or 1.2).

Searching from the chip repo I found some places to read the EnabledTagCompression flag in path but in fact I only saw ir used in tests.

Please tell me if I'm doing/expecting anything wrong, or my data are incorrect or any other error - or if really this feature is not implemented? The latter would mean that this feature is also not implemented in any controller (tested google and chip-tool and both are failing with Tag compression). If not then this feature should be removed from the specs or at least marked as provisional or such!

Here the log from chip tool (receiving side):

```
[1695236295743] [25819:10563642] [EM] <<< [E:13854i S:13287 M:144696026] (S) Msg TX to 5:000000082461914C [CE78] --- Type 0001:02 (IM:ReadRequest)
[1695236295743] [25819:10563642] [IN] (S) Sending msg 144696026 on secure session with LSID: 13287
[1695236295743] [25819:10563642] [DMG] MoveToState ReadClient[0x7f7921408950]: Moving to [AwaitingIn]
[1695236295743] [25819:10563642] [EM] <<< [E:13853i S:0 M:218569596 (Ack:1045704321)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1695236295744] [25819:10563642] [IN] (U) Sending msg 218569596 to IP address 'UDP:[fe80::1c3e:5f63:6021:c8f5%en0]:5542'
[1695236295744] [25819:10563642] [EM] Flushed pending ack for MessageCounter:1045704321 on exchange 13853i
[1695236295744] [25819:10563642] [DL] _OnPlatformEvent
[1695236295744] [25819:10563642] [DL] _OnPlatformEvent
[1695236295779] [25819:10563642] [EM] >>> [E:13854i S:13287 M:1045704322 (Ack:144696026)] (S) Msg RX from 5:000000082461914C [CE78] --- Type 0001:05 (IM:ReportData)
[1695236295779] [25819:10563642] [EM] Found matching exchange: 13854i, Delegate: 0x7f7921408960
[1695236295779] [25819:10563642] [EM] Rxd Ack; Removing MessageCounter:144696026 from Retrans Table on exchange 13854i
[1695236295779] [25819:10563642] [DMG] ReportDataMessage =
[1695236295779] [25819:10563642] [DMG] {
[1695236295779] [25819:10563642] [DMG] 	AttributeReportIBs =
[1695236295779] [25819:10563642] [DMG] 	[
[1695236295779] [25819:10563642] [DMG] 		AttributeReportIB =
[1695236295779] [25819:10563642] [DMG] 		{
[1695236295779] [25819:10563642] [DMG] 			AttributeDataIB =
[1695236295779] [25819:10563642] [DMG] 			{
[1695236295779] [25819:10563642] [DMG] 				DataVersion = 0x367f927b,
[1695236295779] [25819:10563642] [DMG] 				AttributePathIB =
[1695236295779] [25819:10563642] [DMG] 				{
[1695236295779] [25819:10563642] [DMG] 					Endpoint = 0x0,
[1695236295779] [25819:10563642] [DMG] 					Cluster = 0x1d,
[1695236295779] [25819:10563642] [DMG] 					Attribute = 0x0000_0000,
[1695236295779] [25819:10563642] [DMG] 				}
[1695236295779] [25819:10563642] [DMG] 					
[1695236295779] [25819:10563642] [DMG] 				Data = [
[1695236295779] [25819:10563642] [DMG] 						
[1695236295779] [25819:10563642] [DMG] 					{
[1695236295779] [25819:10563642] [DMG] 						0x0 = 22, 
[1695236295779] [25819:10563642] [DMG] 						0x1 = 1, 
[1695236295779] [25819:10563642] [DMG] 					},
[1695236295779] [25819:10563642] [DMG] 				],
[1695236295779] [25819:10563642] [DMG] 			},
[1695236295779] [25819:10563642] [DMG] 			
[1695236295779] [25819:10563642] [DMG] 		},
[1695236295779] [25819:10563642] [DMG] 		
[1695236295779] [25819:10563642] [DMG] 		AttributeReportIB =
[1695236295779] [25819:10563642] [DMG] 		{
[1695236295779] [25819:10563642] [DMG] 			AttributeDataIB =
[1695236295779] [25819:10563642] [DMG] 			{
[1695236295779] [25819:10563642] [DMG] 				AttributePathIB =
[1695236295779] [25819:10563642] [DMG] 				{
[1695236295779] [25819:10563642] [DMG] 					enableTagCompression = true, 
[1695236295779] [25819:10563642] [DMG] 					Cluster = 0x1d,
[1695236295779] [25819:10563642] [DMG] 					Attribute = 0x0000_0001,
[1695236295779] [25819:10563642] [DMG] 				}
[1695236295779] [25819:10563642] [DMG] 					
[1695236295779] [25819:10563642] [DMG] 				Data = [
[1695236295779] [25819:10563642] [DMG] 						29, 31, 40, 48, 49, 51, 60, 62, 63, 
[1695236295779] [25819:10563642] [DMG] 				],
[... 651 lines cut ...]
[1695236295789] [25819:10563642] [DMG] 	
[1695236295789] [25819:10563642] [DMG] 	MoreChunkedMessages = true, 
[1695236295789] [25819:10563642] [DMG] 	SuppressResponse = false, 
[1695236295789] [25819:10563642] [DMG] 	InteractionModelRevision = 10
[1695236295789] [25819:10563642] [DMG] }
[1695236295789] [25819:10563642] [EM] <<< [E:13854i S:13287 M:144696027 (Ack:1045704322)] (S) Msg TX to 5:000000082461914C [CE78] --- Type 0001:01 (IM:StatusResponse)
[1695236295789] [25819:10563642] [IN] (S) Sending msg 144696027 on secure session with LSID: 13287
[1695236295790] [25819:10563642] [TOO] Error: src/app/ReadClient.cpp:643: CHIP Error 0x000000B5: Malformed Interacton Model Attribute Path IB
[1695236295790] [25819:10563642] [DL] _OnPlatformEvent
[1695236295790] [25819:10563642] [DL] _OnPlatformEvent
[1695236295791] [25819:10563635] [CTL] Shutting down the commissioner

```

### Bug prevalence

anytime

### GitHub hash of the SDK that was being used

b46092b69e6556ba7f39a7dd6e83a2afd1ecdb02

### Platform

other

bzbarsky-apple, 2023-09-21:
Yes, this is why https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/7400 got filed in August: "Encoding tag compression needs to be marked provisional".

Nothing implements it right now, and when it _is_ implemented it will need to be conditional on interaction model revision.

That said, it's good to have an SDK issue tracking this too.

Apollon77, 2023-09-21:
Thank you for clarification.