project-chip/connectedhomeip #29363: [CERT-TEST-FAILURE] TC-TIMESYNC-2.2 - TEST FAILURE

Saravana-kr22, 2023-09-21:
### Feature Area

Area:TimeSync

### Test Case

TC-TIMESYNC-2.2

### Reproduction steps

Command to execute the TC-TIMESYNC-2.2 : `python3 TC_TIMESYNC_2_2.py --commissioning-method on-network --discriminator 3840 --passcode 20202021 --storage-path admin_storage.json --PICS ci-pics-values  --paa-trust-store-path /paa_roots`

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

None

### Platform

python, raspi

### Anything else?

TC_TIMESYNC_2_2.py is missing with the  pics  and the test-step 5: TH reads from the DUT the TimeSource (optional) attribute.  

**Reference**:  https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/time_synchronization.adoc#tc-timesync-2-2-setutctime-command-with-dut_server

cecille, 2023-09-21:
Is this the same test as this:
https://github.com/project-chip/connectedhomeip/issues/29343?

cecille, 2023-09-21:
OH! sorry, misunderstood - the script is missing a step.  kk, will fix.

PR #29416 (2023-09-27): TC-TIMESYNC-2.2: Add missing step
Fixes: #29363
closed 2023-09-27 by andy31415