project-chip/connectedhomeip #29368: [Build] Allow exceptions and runtime type information

xvortep, 2023-09-21:
### Build issue(s)

Greetings,

Some SDKs work with exceptions and require rtti. In order for such SDKs to be integrated, flags like -fexception and -frtti have to be set. As far as I can see, there are no methods for setting -fexceptions and -frtti flags other than changing the compiler/BUILD.gn file.

Additionally, rtti flag can be set by providing an additional argument to gn:
 `enable_rtti = true`
which overrides these few lines in compiler/BUILD.gn:
`declare_args() {
  /# Enable Runtime Type Information (RTTI)
  enable_rtti = current_os == "mac" || current_os == "ios"
}`
This method does not sit well with me because of overriding.

However, such mechanism doesn't exist for exceptions, so the only way to set the -fexceptions flag is to change this line in compiler/BUILD.gn:
`config("exceptions_default") {
  configs = [ ":no_exceptions" ]
}`
to:
`config("exceptions_default") {
  configs = [ ":exceptions" ]
}`

Platform: Linux ubuntu 20.04

If there is a way to do this adequately, please correct me.
Thanks.

bzbarsky-apple, 2023-09-21:
>  This method does not sit well with me because of overriding.

This is not "overriding".  This is a build config option (`declare_args` means "this is an argument consumers can set"), which has some default value (enabled on macOS/iOS, disabled elsewhere) but you can set it to whatever you want to get the behavior you want.

@xvortep It seems like the right thing to do here is to add a similar enable_exceptions build config option, right?

closed 2023-09-21 by mergify[bot]

xvortep, 2023-09-25:
> This is not "overriding". This is a build config option (`declare_args` means "this is an argument consumers can set"), which has some default value (enabled on macOS/iOS, disabled elsewhere) but [...]

Sounds right. Thank you.