project-chip/connectedhomeip #29369: [TC-TCTL-3.2] Step 7. SetTemperatureCommand with an invalid value but outcome is unexpected

mideayanghui, 2023-09-21:
### Feature Area

Area:Refrigerators

### Test Case

TC-TCTL-3.2

### Reproduction steps

Expected: Step 7, TH sends command SetTemperatureCommand with an invalid value.
Verify DUT responds Status CONSTRAINT_ERROR(0x87)

Actually, DUT responds Status 0x00.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

 5ccb774e7e5fe7ead3ff85bec1cda31a459c9c37

### Platform

core

### Anything else?

[[TC-TCTL-3.2]_2CHIPTool_log.log](https://github.com/project-chip/connectedhomeip/files/12683103/TC-TCTL-3.2._2CHIPTool_log.log)

jadhavrohit924, 2023-09-21:
@mideayanghui Which features did you enabled? Please read feature map attribute of the cluster.

mideayanghui, 2023-09-22:
The feature map of tempearture control cluster is 5

PR #29409 (2023-09-22): Fix constraint error check for the STEP feature of Temperature Control cluster.
### Problem
- Constraint check was missing for step feature.
- Fixes #29369

### Testing
- Tested cluster control with valid and invalid arguments.
closed 2023-09-22 by mergify[bot]