project-chip/connectedhomeip #29370: [SVE] TC-DGGEN-2.1 inconsistency with test plan

umbraHare, 2023-09-21:
### Reproduction steps

In TC-DGGEN-2.1, the scripts and test plan with eventTrigger are inconsistent and the DUT no longer requires step 9c to trigger an unknown function.

```yaml
    - label:
          "Step 9c: When configured TRUE, Node has been configured with one or
          more test event triggers by virtue of the internally programmed non
          zero EnableKey value"
      runIf: IsExpectedValue
      PICS: DGGEN.S.C00.Rsp
      command: "TestEventTrigger"
      arguments:
          values:
              - name: "EnableKey"
                value: TEST_EVENT_ENABLE_KEY
              - name: "EventTrigger"
                value: TEST_EVENT_TRIGGER_KEY
```

And the script has some descriptive errors at line 479: `Step 10c` should be `Step 10d`

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

5ccb774e7e5fe7ead3ff85bec1cda31a459c9c37

### Platform

raspi

### Type

Spec Compliance Issue

### Testing

Unit tested

bzbarsky-apple, 2023-09-21:
@umbraHare Please report test script issues in https://github.com/CHIP-Specifications/chip-test-scripts/issues

umbraHare, 2023-09-22:
The issue will be moved.

closed 2023-09-22 by umbraHare