project-chip/connectedhomeip #29374: [Tooling]

aravindone, 2023-09-21:
### Reproduction steps / Feature

Hi , i have cloned the latest master branch 

1)activated the matter sdk -success
2)created python env -success
3)runned chip-repl -faced error ,

can you give the suggestion to solve it

### Platform

python

### Platform Version(s)

master branch

### Type

Manually tested with SDK

### Anything else?

[python-repl_error.TXT](https://github.com/project-chip/connectedhomeip/files/12684969/python-repl_error.TXT)

tehampson, 2023-09-21:
I'll take a look next week. I am OOO until then

aravindone, 2023-09-21:
Hi @tehampson thanks for your reply, in the mean time can I have some backup plan that I can use ?

tehampson, 2023-09-25:
Today on my linux machine I sync'd to tip of tree and did the following without running into any issue:
* `scripts/bootstrap.sh`
* `scripts/activate.sh`
* `scripts/run_in_build_env.sh './scripts/build_python.sh --install_wheel build-env -d true' && source ./scripts/activate.sh`
* `chip-repl`

I was able to drop into interactive python repl shell without any issues. @aravindone is that what you are doing? If not can you try that out.

tehampson, 2023-09-25:
Sorry scratch that... I just realized I didn't sync properly and still on an old branch. Give me a few mins to report back after sync it done properly

tehampson, 2023-09-25:
Alright I am able to reproduce the issue. Looking into it now. I'll report back later today with findings

closed 2023-09-25 by tehampson

aravindone, 2023-09-26:
@tehampson Thank you.

PR #29441 (2023-09-25): Correct name of args variable
Fixes: #29374

`argparse` is taking argument `--trust-store` which should be `args.trust_store` and not `args.trustStore`

Test: From a fresh repo
```
source scripts/bootstrap.sh
source scripts/activate.sh
scripts/run_in_build_env.sh './scripts/build_python.sh --install_virtual_env out/venv -d true'
source out/venv/bin/activate
chip-repl
```

Confirm no longer hitting error described in issue