project-chip/connectedhomeip #29375: [CERT-TEST-FAILURE] Test_TC_CADMIN_1_16 expects the reuse of list entry indices, but not required by specs

Apollon77, 2023-09-21:
### Feature Area

Other

### Test Case

Test_TC_CADMIN_1_16

### Reproduction steps

The test is creating 3 fabrics (list: 1,2,3), then deletes the second (list:1,3), then creates a forth fabric but expects that the "just deleted list entry #2" is reused. So expectation is "list: 1,4,3" then.

https://github.com/project-chip/connectedhomeip/blob/82a3e3da613b993e05fe80f70b794c94250c05ca/src/app/tests/suites/certification/Test_TC_CADMIN_1_16.yaml#L303-L307

I did not found any specification requirement that mandate to reuse "list indices". If not (maybe @bzbarsky-apple knows?). If this is not the case then this test case step (13) ned to be adjusted to accep the "chip tool way" (which seems to reuse list items) and also other ways (that usually add new items to the end of the list).

### Bug prevalence

anytime

### GitHub hash of the SDK that was being used

verison of 12.8.2023 (can not recall the exact hash)

### Platform

other

bzbarsky-apple, 2023-09-21:
> I did not found any specification requirement that mandate to reuse "list indices". 

There is not.  Filed https://github.com/CHIP-Specifications/chip-test-scripts/issues/666

Apollon77, 2024-01-20:
Issue still persists in current tests

https://github.com/project-chip/matter-test-scripts/issues/11#issuecomment-1902258757

Apollon77, 2024-01-20:
Ok I was on an old status ... works now!

closed 2024-01-20 by Apollon77